UtilitechAS/amsreader-firmware #553: Voltage and current bar charts show wrong values

fnorlund, 2023-05-02:
Hi,
In release 2.2.16 the values in voltage and current charts seem to be 10 respective 1000 times too high.

[image]

gskjold, 2023-05-02:
Wild guess, you have a Kaifa meter? In config, set multiplier 0.1 and 0.001 respectively

fnorlund, 2023-05-02:
Sorry, I missed there is a config possibility as I didn't have the problem prior to the update.
However, the current multiplier can only be set to min 0.01. Voltage okey.

Thanks for fast response :-)

gskjold, 2023-05-03:
Aha, I see that now. Attached is a firmware with extended range.
Which country is this?

[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11379622/esp32.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11379623/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11379624/esp32s2.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11379625/esp32solo.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/11379626/esp8266.zip)