UtilitechAS/amsreader-firmware #554: Month and day usage are equal

Sommerdal-ufm, 2023-05-03:
**Describe the bug**
The same value are shown for day and month value

**To Reproduce**
1. Main panel
2. Scroll down

**Expected behavior**
Different values except 1. day of month

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Denmark 
 - Meter: Kamstrup 
 - Encryption enabled [yes/no] yes
 - AMS reader: [e.g. Pow-U, ESP32 etc] pow-k+
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [e.g. 2.1.0] 2.1.15/16
 - MQTT: [yes/no] yws
 - MQTT payload type: [e.g. JSON] json
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no] yes

**Additional context**
Add any other context about the problem here.

gskjold, 2023-05-03:
Could you add a screenshot please?

ArnieO, 2023-05-03:
Yes, something is certainly wrong here!
Firmware v2.2.16:
[image]
[image]

gskjold, 2023-05-03:
This is a side effect of issue #552, data for the graph was not updated since 30apr and is unfortunately lost. An average have been set for the three days

Sommerdal-ufm, 2023-05-03:
Here is my observation 
[image]

gskjold, 2023-05-04:
Found the issue, patch will be released after some more testing

closed 2023-05-04 by gskjold