UtilitechAS/amsreader-firmware #555: Use pricing from energi data service for home assistent

Sommerdal-ufm, 2023-05-03:
**Is your feature request related to a problem? Please describe.**
Current and future price is lacking information, which could be obtained from
https://github.com/MTrab/energidataservice

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

gskjold, 2023-05-03:
You need to provide more information on what you think is lacking

ArnieO, 2023-05-03:
@Sommerdal-ufm 
Did you recently bought a Pow-K+? (Guessing from your name). 
If prices are missing, it might just be a configuration issue. 
You need to activate "Enable price fetch", and set your Price region, currency and optionally change the multiplier (I use 1.25 to simulate 25% moms:
[image]

Sommerdal-ufm, 2023-05-03:
Price fetch is enabled and VAT is calculated (1.25) but I lack transport fees and charges that varies during the day.
[image]

Sommerdal-ufm, 2023-05-03:
[image]

Tbnilen, 2023-05-08:
In Denmark, there are different tariffs for the transport of electricity (kWh) during the day:
2400-0600 low load
0600-1700 high load
1700-2400 peak load
The prices change during the summer/winter season (and there is a difference from electricity company to electricity company):

1/4 - 30/9 summer tarif
1/10-31/3 winter half tarif

example Andel Energi:

[image]

ArnieO, 2023-05-08:
The transport tariffs is a highly complex subject that has been discussed in other threads. There are huge differences in tariffication models even within the same country. Our aim is that this firmware should be usable in any country - and that the price estimates should at least be usable in countries covered by Entso-E (for the energy part of the tariffs). 

We have had some fairly good suggestions on how a general setup might be done, but we have not yet gotten to the point where we can prioritize working on it.

gskjold, 2024-06-06:
As nice as it would be to fetch grid costs from APIs, this will not be implemented in the firmware. Use custom pricing configuration added in v.2.3

closed 2024-06-06 by gskjold