UtilitechAS/amsreader-firmware #556: SVG rendering fails for `Energy use` charts (on newly connected device)

johanrd, 2023-05-03:
**Describe the bug**
Bars for consumption is not rendered in the charts `Energy use last 24 hours (kWh)` and `Energy use last month (kWh)`: Upon first glance I thought it was because the device only had been up and running for a couple of hours, but when inspecting the browser console, I found several errors for `attribute transform: Expected number`, found NaN in the context of svg rendering.

**To Reproduce**
Steps to reproduce the behavior:
1. Load dashboard index of a newly connected pow-u (in my case `ams.local`)
2. Open browser console and see a lot of attribute transform errors, and see that the `Energy use` charts have SVGs with NaN values, e.g. `<g class="tick tick-0 tick-undefined" transform="translate(0, NaN)">`

**Expected behavior**
Charts to show consumption data bars, or at least be be resilient towards "no data" / "not a number" in a way that does not break svgs.

**Screenshots**
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: [e.g. Aidon]
 - Encryption enabled [no]
 - AMS reader: [Pow-U ESP32-S2]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [AMS reader v2.2.16]
 - MQTT: [no]
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

**Additional context**
Possibly relevant: I had recently upgraded firmware from v2.2.13, changed 'fixed price' to `1` and changed tariff thresholds.

data.json:
```json
{"im":27712,"om":0,"mf":40,"i":292,"e":0,"ri":0,"re":514,"ic":9332.394,"ec":0.000,"ric":2.150,"rec":2368.737,"u1":239.40,"u2":239.90,"u3":239.60,"i1":0.70,"i2":0.60,"i3":1.00,"f":0.00,"f1":0.00,"f2":0.00,"f3":0.00,"v":3.285,"r":-78,"t":-127.00,"u":5261,"m":127072,"em":1,"hm":1,"wm":2,"mm":0,"me":0,"p":1.00,"mt":1,"ds":2,"ea":{"x":0.0,"p":[0.00,0.00,0.00],"t":2,"h":{"u":0.14,"c":0.14,"p":0.00,"i":0.00},"d":{"u":0.14,"c":0.14,"p":0.00,"i":0.00},"m":{"u":0.14,"c":0.14,"p":0.00,"i":0.00}},"pe":false,"pr":"10YNO-1--------2","pc":"NOK","he":0,"ee":0,"c":1683141851,"a":true}
```

configuration.json:
```json
{"version":"v2.2.16","g":{"t":"Europe/Oslo","h":"ams","s":0,"u":"","p":""},"m":{"b":0,"p":0,"i":false,"d":2,"f":40,"r":0,"e":{"e":false,"k":"00000000000000000000000000000000","a":"00000000000000000000000000000000"},"m":{"e":false,"w":0.000,"v":0.000,"a":0.000,"c":0.000}},"t":{"t":[2,5,10,15,20,25,50,75,150,65535],"h":3},"w":{"s":"int","p":"***","w":19.5,"z":2,"a":false},"n":{"m":"dhcp","i":"","s":"","g":"","d1":"","d2":"","d":true,"n1":"pool.ntp.org","h":true},"q":{"h":"","p":1883,"u":"","a":"","c":"","b":"","m":0,"s":{"e":false,"c":false,"r":false,"k":false}},"p":{"e":false,"t":"","r":"10YNO-1--------2","c":"NOK","m":1.000,"f":1.0000000000},"d":{"s":false,"t":false,"l":4},"i":{"h":{"p":16,"u":true},"a":0,"l":{"p":null,"i":true},"r":{"r":13,"g":14,"b":null,"i":true},"t":{"d":null,"a":null},"v":{"p":10,"o":0.00,"m":1.000,"d":{"v":33,"g":22},"b":0.0}},"u":{"i":1,"e":2,"v":2,"a":2,"r":0,"c":1,"t":2,"p":2,"d":1,"m":1,"s":2},"o":{"e":0,"c":0,"u1":0,"u2":0,"u3":0},"h":{"t":"","h":"","n":""}}
```

gskjold, 2023-05-04:
Thanks, regarding the errors, these always show up because the svg is rendered before it receives the data. I agree this should be fixed, will keep this on the list for next patch!

Have the graph started to populate? I think maybe I see the same error on a device I started up yesterday, will investigate

johanrd, 2023-05-04:
ok, great, thanks.

FYI: the graphs have still not started to populate (in case there are other-than-SVG-bugs to be documented / fixed), though I am still running on `v2.2.16`.