UtilitechAS/amsreader-firmware #557: Deviation between Entso-E price and hourly price per kWh in NOK

ArnieO, 2023-05-05:
**Describe the bug**
The price displayed is wrong by close to 3%

Example:
5th May, hour from 11 to 12, Price region NO1:

- [Entso-E price](https://transparency.entsoe.eu/transmission-domain/r2/dayAheadPrices/show?name=&defaultValue=false&viewType=GRAPH&areaType=BZN&atch=false&dateTime.dateTime=05.05.2023+00:00|CET|DAY&biddingZone.values=CTY|10YNO-0--------C!BZN|10YNO-1--------2&resolution.values=PT15M&resolution.values=PT30M&resolution.values=PT60M&dateTime.timezone=CET_CEST&dateTime.timezone_input=CET+(UTC+1)+/+CEST+(UTC+2)) is 82.06 EUR/MWh.
- Latest [exchange rate from Norges Bank](https://www.norges-bank.no/tema/Statistikk/Valutakurser/) (4tyh May) is 1 EUR = 11,8282 NOK
- Price per kWh should be 82.06 EUR/MWh * 11,8282 NOK/EUR / 1000 kWh/MWh = **0,9706 NOK/kWh**
- Using multiplicator 1,25 (moms) gives 1,21 NOK/KWh
- This user reads 1,18 NOK/kWh:
[image]
[image]

gskjold, 2023-05-05:
Noted. I need to start logging the exchange rate used. Will keep an eye on this

gskjold, 2023-05-05:
Just checked Norges Bank API (which we use) and it returns 11.5265 as exchange rate:
https://data.norges-bank.no/api/data/EXR/M.EUR.NOK.SP?lastNObservations=1&format=sdmx-json

I see that the endpoint I use returns exchange rate for the month, so I should switch to businessday which returns 11.6895:
https://data.norges-bank.no/api/data/EXR/B.EUR.NOK.SP?lastNObservations=1&format=sdmx-json

This will put us closer to the "correct" value

gskjold, 2023-05-05:
Updated amshub for non-apikey clients and updated entso-e code in firmware for next patch for those who use api key

closed 2023-05-05 by gskjold

ArnieO, 2023-05-05:
> I see that the endpoint I use returns exchange rate for the month, so I should switch to businessday which returns 11.6895:
https://data.norges-bank.no/api/data/EXR/B.EUR.NOK.SP?lastNObservations=1&format=sdmx-json

With that exchange rate the above quoted user example would give 1,20 NOK/kWh - which is much closer, but still not identical with Nordpool:
[image]

Which gives:  0,97146 * 1,25 = 1,21 

But I think we can live with differences in the order of 1% due to exchange rate issues.

nossaile, 2023-05-12:
Just for your information: Currently Pow-P1’s hourly prices for SE4 are 0.91% higher than Nordpol’s.