UtilitechAS/amsreader-firmware #559: ESP reports not working

Sommerdal-ufm, 2023-05-06:
**Describe your problem**
ESP is red.
Shows no usage but reports to MQTT.
Other indicators are green.
Time shows 1970.
Unable to firmware update.
Can reboot - but ends in same state.
Can activate configuration 

**Hardware information:**
 - Country: Denmark 
 - Meter: Kamstrup 
 - Encryption enabled yes
 - AMS reader: Pow-K+
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version:. 2.1.16
 - MQTT: yes
 - MQTT payload type: home assistent 
![Screenshot_2023-05-06-08-28-07-64_40deb401b9ffe8e1df2f1cc5ba480b12]
 - HAN baud and parity: 2400 8E1
 - ENTSO-E API enabled: yes
[image]

ArnieO, 2023-05-06:
What you report seems inconsistent with your screenclip (reports to MQTT / Other indicators are green)?
As a start, can you share a screenclip of your config page?

Sommerdal-ufm, 2023-05-06:
[image]

Sommerdal-ufm, 2023-05-06:
Problem solved.
Clearing browser cache did the trick.

closed 2023-05-06 by Sommerdal-ufm