UtilitechAS/amsreader-firmware #560: Negative energy prices break chart layout

kimsey0, 2023-05-06:
The day-ahead prices in [Nordpool DK2](https://www.nordpoolgroup.com/en/Market-data1/Dayahead/Area-Prices/DK/Hourly/?view=table) are currently negative for some hours, and it breaks the chart layout:
[image]

ArnieO, 2023-05-07:
Nice catch - this is the first time we've seen that situation. I agree; the possibility of negative prices should be taken into account in the firmware - and I guess it impacts more than only the barchart.

ArnieO, 2023-05-16:
Just adding input on how this could look - and looking forward to seeing this more! 😅 

But PS: 
Is it negative price that somehow has broken the price fetch? Notice that x-axis labels go to 01 (ending at 23 expected), and no prices for tomorrow yet (plot taken at 18h50).
[image]

gskjold, 2023-05-16:
> Is it negative price that somehow has broken the price fetch?

Fetch isnt broken, but value at 02:00 is -0.0127 which is considered null in the code. After 02:00 prices for tomorrow will come back. I will fix all this ofc :)

gskjold, 2023-05-16:
Or just reboot now, I have made the server return -0.0126  ;)

gskjold, 2023-05-16:
Did some adjustments to handle negative prices better:
[image]

closed 2023-05-16 by gskjold

ArnieO, 2023-05-17:
I happened to be awake around 2 am, and the graph was then fine (sorry, no screen dump). Maybe due to your little fix! 👍