UtilitechAS/amsreader-firmware #561: Missing  values in monthplot after upgrading to 2.2.17

tronde-ams, 2023-05-06:
Norway / aidon 3p IT / ESP32

After upgrading from 2.2.15 (I think) the monthly values from 30 April and after disappeared. The look was OK after ugrade at about 02:00 tonight, but now it looks like this. I have only this unit active at the moment, so I can't compare. I understand there was some issues with the monthplot in one version, but I have not experienced it.

[image]

gskjold, 2023-05-06:
An unfortunate side effect of #552 is that it did not update month plot when entering may, but kept values from the month before. The day consumption from these days are lost and will be averaged when you pass the upcoming midnight

tronde-ams, 2023-05-06:
OK. No big deal then.

closed 2023-05-17 by gskjold