UtilitechAS/amsreader-firmware #562: AMS reader occasionally blanks out for some seconds

kmengvik, 2023-05-08:
Ocassionally the AMS reader blanks out for a few seconds before it is back. Have not yet found any IP conflict on the router, so can not say that the problem is there..

**Screenshots**
[image]

**Hardware information:**
 - Country: [Norway]
 - Meter: [Kamstrup]
 - Encryption enabled [NO]
 - AMS reader: [Chip: esp32s2
Device: [Pow-K+]

**Relevant firmware information:**
 - Version: [e.g. 2.1.17]
 - MQTT: [yes]
 - MQTT payload type: [RAW (full)]
 - HAN GPIO: [?]
 - HAN baud and parity: [e.g. 2400 8E1]
  - ENTSO-E API enabled: [no]

ArnieO, 2023-05-09:
We've seen 2 or 3 such cases before. 

At least one case turned out to be a (rather intricate) IP conflict. 
If you use static IP: Could you try to change it and see if situation improves?

gskjold, 2023-05-17:
Closed due to inactivity

closed 2023-05-17 by gskjold