UtilitechAS/amsreader-firmware #563: PriceAPI: Unspecified error -1

wdpk88, 2023-05-08:
Getting this error daily. Goes away when i reboot. But comes back after a few hours. Seams the price graph screws up to. Before and after reboot:

[image]
[image]

ArnieO, 2023-05-09:
Huh. Do you use Entso-E API key (or do you read prices from our intermediate server)?

wdpk88, 2023-05-09:
Using a API Key

[image]

gskjold, 2023-05-09:
Are you using DHCP? Could you please check DNS servers on the status page?

wdpk88, 2023-05-10:
[image]

ArnieO, 2023-05-10:
I too seem to have issues today:
[image]

[image]

[image]
Rebooting did not help.

gskjold, 2023-05-10:
@wdpk88 Could you go to the status page and confirm that your static DNS servers are indeed applied?
[image]

wdpk88, 2023-05-10:
[image]

gskjold, 2023-05-10:
Yes, but also please check the actual configuration state
[image]

wdpk88, 2023-05-10:
Sorry :)

[image]

gskjold, 2023-05-10:
This aligns with feedback from a different customer, who is actually running DHCP. His DNS config has also been overwritten, but with 32.1.70.0 . I have not yet been able to identify where this corruption comes from, but I will continue searching

ArnieO, 2023-05-11:
> This aligns with feedback from a different customer, who is actually running DHCP. His DNS config has also been overwritten, but with 32.1.70.0 . I have not yet been able to identify where this corruption comes from, but I will continue searching

Interesting. So we have reports on similar DNS corruption both with DHCP and static IP?

gskjold, 2023-05-11:
> Interesting. So we have reports on similar DNS corruption both with DHCP and static IP?

Correct. Most likely caused by a memory corruption from somewhere in the code

tha1971, 2023-05-13:
Hi
My dns is also getting changed after some hours ( in this case my mqtt crashed after approximately 6:45 hours)
[image]
[image]

ArnieO, 2023-05-13:
So DNS was correct but was corrupted during operation?

tha1971, 2023-05-13:
It seams that the DNS was corrupted during operation and a reboot corrected it again

gskjold, 2023-05-13:
The reason for this is still a mystery to me... I am wondering if this is related to how the device is configured. For those who have issues, it would be nice if you could share your full configuration with me (but without secrets). You can send it to post@utilitech.no if you prefer not to share it here.

tha1971, 2023-05-15:
For me it looks to go down around the same time (app 1300) as it is supposed to get the new day ahead prices from entsoe

gskjold, 2023-05-15:
Thanks, that's good information to have!

gskjold, 2023-05-17:
Finally able to reproduce this after moving the device to my non-IoT WiFi network. I have been chasing this for hours, and boiled down the problem to a test code here: https://github.com/gskjold/esp32-wifi-problem
No idea how to fix this, so I have opened an issue with espressif: https://github.com/espressif/arduino-esp32/issues/8221

gskjold, 2023-05-18:
Some new information could indicate that the issue is related to IPv6. Do any of you know if you have IPv6 enabled on your network? If you by any chance have Telenor as your ISP, the answer is yes. Please let me know.

tha1971, 2023-05-18:
Hi
My isp is Stofa.
But they did some work on the network in my city recently.
I can check the ipv6 to night

wdpk88, 2023-05-18:
I have Telenor as ISP

gskjold, 2023-05-21:
Added workaround, new patch soon

closed 2023-05-21 by gskjold

oyvindhn, 2024-02-20:
Updated to from v2.2.24 to v2.2.28 today and lost all prices.  Errormessage:  PriceAPI: Connection error

A know fault?

ArnieO, 2024-02-20:
> Updated to from v2.2.24 to v2.2.28 today and lost all prices. Errormessage: PriceAPI: Connection error
> 
> A know fault?

No. 
But due to often seen issues with Entso-E we recommend not using the Entso-E API key. Instead, leave the field open. The device will then fetch prices from our intermediary server. It is a simpler API call, and we attempt alternative price sources if Entso-E is not available/responsive.