UtilitechAS/amsreader-firmware #565: Follow up of issue #528

spinehead, 2023-05-11:
**Describe the bug**
Just a follw up on an earlier case. 
There are still issues with display of consumtion values in the user interface. 
Please have a look at the screenshots.
With multilplier for kWh set to 0, the meter stand is displayed correctly (78xx,x) 
I have tested with multipliers of 10 and 100, but result is that accumulated consumtion data (hours, time and month) becomes completely wrong
I have connected (via MQTT) to Home Assistant and the consumtion values shown here is correct (see picture below)

**To Reproduce**
Steps to reproduce the behavior:
Nothing special is requied to reproduce the issue. It's constant

**Screenshots**
[image]
[image]
[image]
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled no
 - AMS reader: Pow-U
 - M-bus adapter (if applicable):

**Relevant firmware information:**
Version: 2.2.17.
MQTT: Yes
MQTT payload type: Home Assistant
HAN GPIO: [e.g. GPIO5]
HAN baud and parity: 2400 8E1
Temperature sensors NA
ENTSO-E API enabled: yes

**Additional context**
Add any other context about the problem here.

gskjold, 2023-05-11:
Looking at #528 I see that your meter reading then showed 7739.4kwh (18apr) and now you have 7822.7kwh (11may). Considering your 400A main fuse, I have my doubts that you have only used 83.3kwh in almost a month. This leads me to believe that your multiplier is wrong. You need to find the correct multiplier that gives you the same reading on the dashboard that you have on your meter. That same multiplier should most likely be used for watts and amps.

Swapping multipliers does not change the data in the graph, it will only apply to new data points, so you will need to wait a few hours to see how that new multiplier looks. But the best way is to compare the reading on the dashboard with the reading on your meter and choose the multiplier from that.

gskjold, 2023-05-16:
Excellent! Thanks for reporting back to us