UtilitechAS/amsreader-firmware #566: A new parameter is needed to calculate hourly cost

nossaile, 2023-05-12:
In Sweden (and probably in other countries as well) the hourly cost is calculated as follows:
 HourlyCost = (SpotPrice ENTSO-E + TransferFee + EnergyTax) * (1 + moms) 

Thus a parameter "Addition" is needed, enabling AMS Reader to calculate

HourlyCost = (SpotPrice ENTSO-E + Addition) * Multiplier

The importance of Addition is underline by the fact that in Sweden SE4 the TranferFee is SEK 0.2116/kWh and the EnergyTax is SEK 0.3920/kWh, it’s a large part of the monthly energy bill.

gskjold, 2023-05-12:
Technically duplicate of #252

closed 2023-05-12 by gskjold

nossaile, 2023-05-12:
I protest!!! This is needed for every user you have in Sweden and it's not a duplicate of #252! Without Addition you can take away Multiplier.

gskjold, 2023-05-12:
See comment https://github.com/UtilitechAS/amsreader-firmware/issues/252#issuecomment-1076116363 in that issue, which suggests formula based implementation of price calculation. This would make the calculation extremely dynamic.

All of Europe have the issue you are describing, the next country more complex than the other.