UtilitechAS/amsreader-firmware #569: FW uppgrade error

sm4upt, 2023-05-15:
**Describe your problem**
FW uppgrade from v.2.2.15 to v2.2.17 fails with following error message.
Previous upgrade attempt (v2.2.17) does not match current version (v2.2.15) [0/-1]

**Hardware information:**
 - Country: [Sweden]
 - Meter: [Kamstrup 6841131BN245101092]
 - Encryption enabled [no]
 - AMS reader: ( Pow-K+, ESP32s2]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [2.2.15]
 - MQTT: [yes]
 - MQTT payload type: [HomeAssistant]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [2400 8N1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [no]

gskjold, 2023-05-15:
Please check on status page if your DNS server have been corrupted (See issue #563). If that is the case, could I get a full config file from you? post@utilitech.no
[image]

gskjold, 2023-05-15:
Excellent, this is the same problem as #563 then. Keep an eye on #563 and upgrade when that is fixed.

closed 2023-05-21 by gskjold