UtilitechAS/amsreader-firmware #573: Future energy price value

Nealux, 2023-05-22:
When hoovering with the cursor above the staples representing the hourly prices you see a more exact figure (nice feature).
But when I get to hour 2 and later this does not work. No exact figure is displayed.
This was also a problem with the 2.2.18 release
This morning before the new prices were loaded the exact price was displayed  for all available hours (as I recall)!

Steps to reproduce the behavior:
1. Go to future energy price 
2. Click on hour 02 or later
**Expected behavior**
Exact price displayed for every hour.

**Hardware information:**
 - Country: [Sweden
 - Meter:Sagemcom
 - Encryption enabled no
 - AMS reader: Prototype
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version:2.2.19]
 - MQTT: no
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity:115000 8n1]yes[yes/no]

ArnieO, 2023-05-24:
It looks fine here!
Can you provide a screenshot example, please?

ArnieO, 2023-05-24:
If you use the delay function of Snipping tool you can capture the mouseover.
Here is my mouseover at 23h, showing correct value:
[image]

Nealux, 2023-05-24:
Ia have screen shoots now from 01 and 02 that they dosen't really say much as the curzor is not captured..
But the problem shows up after the price has been updated for today and I hoover over hour 02:00 and all hours  that follows.
This is true today.
But tomorrow it will show the exact price until the price is updated tomorrow.
Yes strange but that is the way it is...
Screen shot with corzor hoovering over 01:00
[image]
This screen shoot with the curzor above 02:00
[image]

ArnieO, 2023-05-24:
There might be a (strange; as you point out) problem here - but sorry: I fail to understand what you're seeing when you hover over 02. Maybe easier to make a photo with your phone?

Nealux, 2023-05-24:
I do not see the information as shown in the first example. (-0.04 SEK). This is the information displayed when the cursor is place above the 01.00 o'clock staple.
When the cursor is is moved to the 02:00 staple no information is displayed.
This goes for all staples following the 02:00 staple.
But if I do the same thing in the morning all information is displayed.
It will work ok until the figures are updated  tomorrow.
I'm sorry that  I'm not able to explain what I see as a problem in a better. way.
[image]
[image]

ArnieO, 2023-05-25:
OK, I finally understood; sorry that I was a bit slow! I was confused by your first message saying " No exact figure is displayed." 
I thought the problem was precision on the figure, and didn't understand that it was missing totally. 

Thank you for reporting - we'll see if we can find the issue.

cmock, 2023-05-26:
FWIW, I'm having the same issue, and when I inspect the HTML in the browser I see for each bar:

```
<g>
 <rect ...></rect>
 <text ...></text>
 <title>0.10 EUR</title>
</g>
```

But for the bars after 01 hrs, the whole `<title>` tag is missing.

closed 2023-06-11 by gskjold