UtilitechAS/amsreader-firmware #574: Store `Price current hour` in Home Assistant's Long Term Statistics

kandsten, 2023-05-22:
Home Assistant has a facility for storing [Long Term Statistics](https://data.home-assistant.io/docs/statistics/); these live on indefinitely, as opposed to the recorder data that is kept for a finite number of days.

It'd be nice it the energy price that AMS reader publishes would be stored in this database as it enables other interesting uses of the data. One concrete such example: I'm the author of [a heatmap card](https://github.com/kandsten/ha-heatmap-card), looking to expand it to incorporate energy usage and price in the same heat map. The more that works out the box, the better for the user. 
 
One way to achieve it would be to modify the `Price current hour` topic data slightly:

**Current:**
```
unit_of_measurement: NNN/kWh
device_class: monetary
friendly_name: Price current hour
```

**Proposed:**
```
unit_of_measurement: NNN/kWh
state_class: measurement
friendly_name: Price current hour
```

Setting the `state_class` would make it fit [the requirements](https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics) for getting stored into Home Assistant's LTS.

_Ideally_ it'd be possible to keep both `device_class: monetary` and _also_ set `state_class: measurement`, _but_ HA assumes that Monetary is only ever going to be going to be a [Total](https://developers.home-assistant.io/docs/core/entity/sensor/#entities-not-representing-a-total-amount) type value, never a Measurement; this is also enforced in code. This would be the pragmatic way around that.

MrDemocracy, 2023-08-11:
I think this could be why I'm receiving this warning and why not all the statistics entities get listed on the energy dashboard:
[image]
[image]

gskjold, 2023-08-31:
I see I have not commented on this before, sorry about that. I will have a closer look at a few HA related issues in the near future and have this sorted out.

gskjold, 2023-12-15:
Apparently I did this for 2.2.22 :)

closed 2023-12-15 by gskjold