UtilitechAS/amsreader-firmware #575: Price info

PerBuch54, 2023-05-23:
After the latest update i dont get the Future energy price Future energy price updates.

gskjold, 2023-05-23:
If you get a price not found right now, its just entsoe missing price for tomorrow

PerBuch54, 2023-05-24:
You are right prici info ok now

PerBuch54, 2023-05-28:
Again! Are ENTSO-E a little unstable ?

gskjold, 2023-05-28:
Sadly they have had several cases of missing data lately, including today...

nossaile, 2023-05-28:
...and today too. It looks like ENTSO-E is more and more unreliably. Isn’t there any alternative supplier of spot prices?

tha1971, 2023-05-28:
Are #580 the same problem as this, i am also missing prices with wrong date and correct time

nossaile, 2023-05-28:
No, tha1971, this is about missing spot prices and #580 is about wrong date shown.

gskjold, 2023-05-29:
Fallback implemented

closed 2023-05-29 by gskjold