UtilitechAS/amsreader-firmware #576: Energy use last month (kWh) on graph is actually not last month, but last 30 days.

joloevaa, 2023-05-24:
**Describe the bug**
Energy use last month (kWh) is wrong. It's better with: Energy used last 30 days (kWh)

**To Reproduce**
Steps to reproduce the behavior:
1: Look at the graph for energy use.

**Expected behavior**
I expected: Energy used last 30 days (kWh)

**Screenshots**
[image]

**Hardware information:**
 - Country: [e.g. Norway]
 - Meter: [e.g. Aidon]
 - Manufacturer: Aidon
 - Model: 6534

ArnieO, 2023-05-24:
I agree, the title is a bit misleading.

gskjold, 2023-05-24:
It isn't always 30 days though. It is always date to date one month:
[image]

ArnieO, 2023-05-24:
> It isn't always 30 days though. It is always date to date one month: [image]

Right - and I think that is a useful way to display it. 

**Does anyone have a good proposition for a better title?**
The term "previous month" could maybe be read as "previous month", which is incorrect. 
And as the duration of a month is not a fixed figure, so "last 30 days" will also be imprecise. 
Anyone?

gskjold, 2023-05-24:
Technically we know how many bars are in the graph, so we could show "last 28 days" etc depending on the number of bars in the graph. That would mean the number would change month to month, but maybe not a problem? I guess it could be better than "last month", the confusion on that one is understandable. Ideally I would prefer a term that would cover a date-to-date period though :)

ArnieO, 2023-05-24:
Yeah... maybe "Last xx days" is the best option?
Anyone who wants to weigh in here?

henriklund, 2023-05-30:
IMHO I do not see this an huge problem (especially due to how the graphs are shown). Nevertheless, same issue would apply to the 'Energy use last 24 hours (kWh)' graph. If clarity is sought why not simply change 'last' to 'the last'? And you could even clarify this further by a 'the last one month' ?

gskjold, 2023-06-11:
Going for "last 28/29/30/31 days", where the number depends on how many days last month had.

closed 2023-06-11 by gskjold