UtilitechAS/amsreader-firmware #578: Kamstrup Omnia e-meter support in Sweden

kirilldem, 2023-05-26:
**Describe your problem**
I am getting a new Kamstrup Omnia e-meter installed in Sweden and I would like to see if it is compatible with AMSreader and what hardware is needed to self-build the reader.

**Hardware information:**
 - Country: Sweden
 - Meter: Kamstrup Omnia e-meter
 - Encryption enabled 
 - AMS reader: 
 - M-bus adapter (if applicable):

The hardware needs to be compatible with the DSMR 5.02 standard and the Energy Companies' recommendation v2 0-201912.
 The customer interface is based on the Dutch standard DSMR with an RJ12 connector and follows Energiföretagen Sweden's recommendation.

```
Kamstrup OMNIA® e-meter is available with a home area network (HAN) interface accessible on the meter front.
The HAN interface connector type is RJ12 and the meter holds a female connector; the external device connects via
standard RJ12 male plug. The external device is power supplied from the meter (+5V). This physical interface conforms to the properties defined in DSMR v. 5.0.2 P1 companion standard, with the following exception:
• The HAN interface fulfills the electrical requirements of overvoltage category III, which is the level generally applicable to electricity smart meters.
Some functionality described in the DSMR P1 companion standard might not be available. Please refer to the OMNIA®
e-meter user guide for details on available functionality. Note: The meter is available in a version without the +5V
power supply output to the external device. This is selected at ordering.
Due to the typical use cases and processes of distribution system operators (DSOs), the HAN interface is unencrypted. 
```
https://www.goteborgenergi.se/kundservice/nya-smarta-elmatare/sa-fungerar-din-elmatare 

I would like to use D1 Mini pro (for the external antenna) suggested in the hardware on your wiki, but not sure if I need the TSS721 MBUS reader as well? The energy company writes this in their FAQ:
`The electricity meter doesn’t support our remote reading with M-bus outputs. What can we do?
It is correct that the M-bus output is not compatible with the new electricity meter`

I guess I also need some jumper wires and an RJ12 cable to connect to the hardware?

Any help is greatly appreciated.

kirilldem, 2023-05-26:
Having looked into this further, would this https://frogshop.se/product/kort-for-han-p1-ttl-serieinterface-monterat/ together with some jumper wires and esp work with AMS reader?

ArnieO, 2023-05-26:
Let me start by saying that the [Pow-P1](https://amsleser.no/hjem/10-1-pow-p1.html) that we sell will work fine on your Kamstrup Omnia! 😉 

Forget about M-bus, your meter does not have that. It has an Open collector output, with data rate 115 200 baud. 

The adapter you link seems to do what is needed to interface to a 3,3V microcontroller. 
Please note that many users struggle with various Wemos D1 mini clones, for various reasons. If you do a search in the issues here for Wemos D1; you'll see what I mean and the advise I have given to try to assist. Proper powering and solid decoupling is the main issue.

If I were you I would go for a module with ESP32 (S2 or C3), if you find one with external antenna. Those are much more modern, powerful and performant microcontrollers. Our firmware still runs on ESP8266, but it is a tight squeeze! There could in future be features coming that will not be available for devices using ESP8266.

kirilldem, 2023-05-26:
Thanks for your feedback and explanation - makes sense, my reasoning for an ESP8266 was the lower power budget - but based on your suggestion it looks like this board is what I am looking for https://www.aliexpress.com/item/1005004541456510.html 

If you don't see any issues with the board I linked, then I will order it along with the RJ12 to TTL daughterboard I linked earlier and then I will wait for the new meter to drop :)

ArnieO, 2023-05-26:
There is 250 mA available from the P1 port, so power should not be an issue with the ESP32's I proposed. We use ESP32-S2-MINI in Pow-P1. 

That Lilygo board looks nice, but I assume you need to move/rotate this 0R resistor to use the external antenna connector in stead of the onboard 3D antenna:
[image]
[image]
It is quite normal for this kind of boards, that you need to move a 0R resistor to use the IPEX connector. The biggest issue is that this is a very small component; looks like 0402. But it is doable with a good tweezer, a steady hand and good magnification!

Useful resources:
https://www.lilygo.cc/products/t8-c3
https://github.com/Xinyuan-LilyGO/T8-C3

Unlike many Wemos boards, there seems to be a reasonable amount of capacitance on this board, so you have a good hope of not running into voltage stability issues where you need to add capacitance on the 3,3V. 
https://github.com/Xinyuan-LilyGO/T8-C3/blob/main/Schematic/T8-C3_V1.1.pdf

Good luck with your project! 👍

ArnieO, 2023-05-26:
NB! 
Not 100% sure, but you might also need this one to use the card:  https://www.lilygo.cc/products/t-u2t?_pos=1&_psq=t-u2t&_ss=e&_v=1.0

When I studied the schematic for the Lilygo board I did not find the two 5k1 resistors on CC1 and CC2 of the USB-C connector that are normally needed to negotiate correct voltage from a USB-C power. I find one in the schematic of this module (could mean that only one orientation of the USB-C plug works!?), so I would buy this one too, to be sure. 

You get it bundled with the board  if you buy here: https://www.aliexpress.com/item/1005004513872582.html

closed 2023-05-29 by gskjold

kirilldem, 2024-04-06:
I have managed to get readings from the meter, but I am wondering if the meter model should be identified by amsreader?
[image]
Despite this, the values are updated correctly.

My settings are as follows:
[image]

I am on 2.3.1.

ArnieO, 2024-04-07:
That is strange. The meter model should be part of the payload, and be showed like this:
[image]

Please run a [Telnet debug](https://www.amsleser.no/module/ets_blog/blog?id_post=24) in Verbose mode and post what you see for one payload as text here. Attached text file would be greatd.

kirilldem, 2024-04-07:
Here is one payload from verbose telnet debug:

> (V) DSMR frame:
(V) 2F 4B 41 4D  35 0D 0A 0D  0A 30 2D 30  3A 31 2E 30
(V) 2E 30 28 32  34 30 34 30  37 31 32 32  38 35 34 53
(V) 29 0D 0A 31  2D 30 3A 31  2E 38 2E 30  28 30 30 30
(V) 30 39 36 32  31 2E 39 38  33 2A 6B 57  68 29 0D 0A
(V) 31 2D 30 3A  32 2E 38 2E  30 28 30 30  30 30 30 30
(V) 30 30 2E 30  30 30 2A 6B  57 68 29 0D  0A 31 2D 30
(V) 3A 33 2E 38  2E 30 28 30  30 30 30 30  30 30 30 2E
(V) 39 31 36 2A  6B 56 41 72  68 29 0D 0A  31 2D 30 3A
(V) 34 2E 38 2E  30 28 30 30  30 30 32 34  39 36 2E 37
(V) 37 39 2A 6B  56 41 72 68  29 0D 0A 31  2D 30 3A 31
(V) 2E 37 2E 30  28 30 30 30  31 2E 34 35  33 2A 6B 57
(V) 29 0D 0A 31  2D 30 3A 32  2E 37 2E 30  28 30 30 30
(V) 30 2E 30 30  30 2A 6B 57  29 0D 0A 31  2D 30 3A 33
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 30  30 2A 6B 56
(V) 41 72 29 0D  0A 31 2D 30  3A 34 2E 37  2E 30 28 30
(V) 30 30 30 2E  33 32 31 2A  6B 56 41 72  29 0D 0A 31
(V) 2D 30 3A 32  31 2E 37 2E  30 28 30 30  30 31 2E 32
(V) 31 38 2A 6B  57 29 0D 0A  31 2D 30 3A  34 31 2E 37
(V) 2E 30 28 30  30 30 30 2E  32 31 33 2A  6B 57 29 0D
(V) 0A 31 2D 30  3A 36 31 2E  37 2E 30 28  30 30 30 30
(V) 2E 30 32 32  2A 6B 57 29  0D 0A 31 2D  30 3A 32 32
(V) 2E 37 2E 30  28 30 30 30  30 2E 30 30  30 2A 6B 57
(V) 29 0D 0A 31  2D 30 3A 34  32 2E 37 2E  30 28 30 30
(V) 30 30 2E 30  30 30 2A 6B  57 29 0D 0A  31 2D 30 3A
(V) 36 32 2E 37  2E 30 28 30  30 30 30 2E  30 30 30 2A
(V) 6B 57 29 0D  0A 31 2D 30  3A 32 33 2E  37 2E 30 28
(V) 30 30 30 30  2E 30 30 30  2A 6B 56 41  72 29 0D 0A
(V) 31 2D 30 3A  34 33 2E 37  2E 30 28 30  30 30 30 2E
(V) 30 30 30 2A  6B 56 41 72  29 0D 0A 31  2D 30 3A 36
(V) 33 2E 37 2E  30 28 30 30  30 30 2E 30  30 30 2A 6B
(V) 56 41 72 29  0D 0A 31 2D  30 3A 32 34  2E 37 2E 30
(V) 28 30 30 30  30 2E 30 31  39 2A 6B 56  41 72 29 0D
(V) 0A 31 2D 30  3A 34 34 2E  37 2E 30 28  30 30 30 30
(V) 2E 32 31 37  2A 6B 56 41  72 29 0D 0A  31 2D 30 3A
(V) 36 34 2E 37  2E 30 28 30  30 30 30 2E  30 38 35 2A
(V) 6B 56 41 72  29 0D 0A 31  2D 30 3A 33  32 2E 37 2E
(V) 30 28 32 33  37 2E 34 2A  56 29 0D 0A  31 2D 30 3A
(V) 35 32 2E 37  2E 30 28 32  34 31 2E 34  2A 56 29 0D
(V) 0A 31 2D 30  3A 37 32 2E  37 2E 30 28  32 34 31 2E
(V) 34 2A 56 29  0D 0A 31 2D  30 3A 33 31  2E 37 2E 30
(V) 28 30 30 35  2E 31 2A 41  29 0D 0A 31  2D 30 3A 35
(V) 31 2E 37 2E  30 28 30 30  31 2E 34 2A  41 29 0D 0A
(V) 31 2D 30 3A  37 31 2E 37  2E 30 28 30  30 30 2E 34
(V) 2A 41 29 0D  0A 21 31 44  36 36 0D 0A
(D) Received valid DSMR at 0

[image]

And here is a web page with some more potentially useful information https://www.goteborgenergi.se/kundservice/elmatarbyte/sa-fungerar-din-elmatare

ArnieO, 2024-04-07:
OK, so it is an Omnia E-meter with P1-port you have (and not an Omnipower). Sorry, I forgot to consider the rest of the thread in my previous response. 

Thank you for debug output, let us se what @gskjold gets out of that.

gskjold, 2024-04-07:
This hex is straight up ascii, so any hex to ascii converters will give you the content, which unfortunately does not contain meter type or id: 
```
/KAM5

0-0:1.0.0(240407122854S)
1-0:1.8.0(00009621.983*kWh)
1-0:2.8.0(00000000.000*kWh)
1-0:3.8.0(00000000.916*kVArh)
1-0:4.8.0(00002496.779*kVArh)
1-0:1.7.0(0001.453*kW)
1-0:2.7.0(0000.000*kW)
1-0:3.7.0(0000.000*kVAr)
1-0:4.7.0(0000.321*kVAr)
1-0:21.7.0(0001.218*kW)
1-0:41.7.0(0000.213*kW)
1-0:61.7.0(0000.022*kW)
1-0:22.7.0(0000.000*kW)
1-0:42.7.0(0000.000*kW)
1-0:62.7.0(0000.000*kW)
1-0:23.7.0(0000.000*kVAr)
1-0:43.7.0(0000.000*kVAr)
1-0:63.7.0(0000.000*kVAr)
1-0:24.7.0(0000.019*kVAr)
1-0:44.7.0(0000.217*kVAr)
1-0:64.7.0(0000.085*kVAr)
1-0:32.7.0(237.4*V)
1-0:52.7.0(241.4*V)
1-0:72.7.0(241.4*V)
1-0:31.7.0(005.1*A)
1-0:51.7.0(001.4*A)
1-0:71.7.0(000.4*A)
!1D66
```

kirilldem, 2024-04-18:
Thanks, so I guess it works as expected :)