UtilitechAS/amsreader-firmware #583: Better display of prices when close to zero NOK?

ThorAlex87, 2023-05-30:
**Is your feature request related to a problem? Please describe.**
Not really a big problem, but I feel like it could be improved. Prices in NOK are rounded to a single number after the decimal point,  which makes for a not so great graph when many of the future hours are close to zero. The tooltip displays the exact price, but does not help to give a great overview.

**Describe the solution you'd like**
Not sure.

**Describe alternatives you've considered**
Maybe display two digits after the decimal point? could maybe be conditional for prices under 0.1? 

**Additional context**

ThorAlex87, 2023-05-31:
I'm back to one digit too. I suspect the threshold is when there is a price above 0.05 it goes single digit, so maybe raising it up to 0.2 would avoid these situations where most prices are displayed as 0.0.

ArnieO, 2023-05-31:
My highest now is less than 0.14 - so I think it is something else. Something for @gskjold to look into for next revision I guess.

gskjold, 2023-05-31:
Two digits area used when there are less than 20 bars in the plot

ThorAlex87, 2023-05-31:
Is that really necessary? On smaller screens the numbers flip vertically, so there should not be any space issues when there are many bars?

closed 2023-06-11 by gskjold