UtilitechAS/amsreader-firmware #584: Pow-P1 can't boot anymore without power from USB

syska, 2023-06-01:
Today I noticed that the Pow-P1 wasn't reporting data anymore to HA.

And it just looked like this: https://github.com/UtilitechAS/amsreader-firmware/assets/317770/84811d62-f188-430b-b7cc-1ef8c2fbb023

Took it off, powered it with USB from my computer ... worked fine, started up, acccessed the web interface ... 

Flashed it from 2.2.17 to 2.2.19 ... no issues.

Moved it back ... same issue with the same flashing LEDs.

Now I have powered it with a USB adapter and it works.

So how do I debug it from here? Is this a broken device? RMA case?

ArnieO, 2023-06-02:
Thank you for reporting, and for the video. 

This looks like a reboot loop, and it could in fact be due to failing hardware. I see no firmware-related reason it should reboot while powered from the meter, then work correctly with USB power connected.

Are you equipped to measure voltages on the board when it is powered only from the meter?
These are the measurement points:
[image]
Please measure both 3,3V and 5V against GND. Oscilloscope measurement would be great, but if there is an issue you should be able to see it with a multimeter.

If there is a problem on the board we will of course replace it, in that case contact us on post@amsleser.no.

syska, 2023-06-02:
2

> Thank you for reporting, and for the video.
> 
> This looks like a reboot loop, and it could in fact be due to failing hardware. I see no firmware-related reason it should reboot while powered from the meter, then work correctly with USB power connected.
> 
> Are you equipped to measure voltages on the board when it is powered only from the meter? These are the measurement points: [image] Please measure both 3,3V and 5V against GND. Oscilloscope measurement would be great, but if there is an issue you should be able to see it with a multimeter.
> 
> If there is a problem on the board we will of course replace it, in that case contact us on [post@amsleser.no](mailto:post@amsleser.no).

Only got a multimeter and got:
4,93 on the 5,0V
3,28 on the 3,3V

Do note this is a cheap multimeter from the local hardware store.

Funny thing ... after having messed around with it, it looks like it booted from power from the meter only ... which is very strange ... but it gave me a few red blinks before started. ( I think ... it's sometimes hard to tell what these colors are from the small LED's )

I can now access it from the IP over wifi and it sends data to HA.

It seem to be blinking blue(data from the meter) and green(valid data) ... but alot more blue, than green blinks. As I said in the first post ... I'm do get alot of `HAN Footer checksum error` which I guess is the reason I see way more blue than green blinks.

Not sure if the energy company could have changed anything in the meter that could have caused this and then reverted it ... 🤔 

I will keep you updated if this happens again ...

ArnieO, 2023-06-03:
> Only got a multimeter and got:
> 4,93 on the 5,0V
> 3,28 on the 3,3V

OK, so the voltages look fine. 
Multimeter measurement is fine, and the cheap ones are accurate enough. We would just potentially get some more information (noise etc) if you had an oscilloscope.

The checksum error is a potential clue, since that is an other unusual issue on your device. 

So... time to admit a small design flaw on this board:
The original value of R1 was too large, so we make a small modification to each of the boards before testing and flashing/programming. If you look at R1 you will se that it is double; I have soldered a resistor chip on top of the factory installed one to increase the strength of it, acting as pullup resistor for the data input (which is an Open Collector output. Verification of the soldering is done by visual inspection, but I cannot rule out that it might not be perfect. If that soldering is a "near short" or "sometimes almost short", it **could** potentially be a root cause for what you see.

The soldering is a bit tricky to get nice, as it results in a mixture of lead free (original on the board) and leaded solder (added my be). So it always looks a bit crappy, like this random example from a board ready for shipment:
[image]

The fix could have been done by removing the original chip, cleaning the pads to remove the lead free solder, solder in a new chip with the right value. But it is a lot more work - and quite tedious. And there is also a risk of applying too much heat to the pads during this process - which could (also) make the board less reliable. And the fix we do has so far worked well for all boards sold.

IF this soldering of your board has a flaw that I did not detect visually, and it passed the testing (which has limited ability to detect error on this component)... it could potentially be the reason for what you experience.

IF the result is close to being a short, or an intermittent short, the board could draw too much current - potentially causing the reboot loop. It could at the same time affect the reading of data - causing checksum error / many rejected datagrams.

I would love to have this board back for testing, but sending returns across borders is a customs nightmare, so I think we let that pass. 

Please keep me updated, and if this board continues to behave badly, please contact me on post@amsleser.no to get you identified so I can send you a new board!

syska, 2023-06-07:
@ArnieO 

I was like ... maybe I can fix that soldering, until I actual got the board in my hand and saw how freaking small it was.

Sadly, it keeps acting strange ... 

Can't boot from `meter power`. But if I power it up from `USB power` and about 30 seconds later disconnect it ... it can run fine on the `meter power`... ( Yes, it worked this time for about 5 mininutes, then rebooted again and got into a loop.

Also the `HAN: Footer checksum error` worries me.

I will send an mail to post@amsleser.no 

Thank for all the very good support here.

ArnieO, 2023-06-07:
> I will send an mail to [post@amsleser.no](mailto:post@amsleser.no)
> 
> Thank for all the very good support here.

OK, sorry to hear this - but we will of course send you a replacement. 
I just received your email, we'll take it from there.

syska, 2023-07-03:
No luck with the new device. Same issue ... I have no idea what is going on... 

Since I have 230v power close to the device, I'm now just powering it from USB. I can live with that. No reboot in the last 11 days... 

So other than the `HAN: Footer checksum error` ... it's all working really good.

Here is the config i'm using if that could cause the issue:
```
amsconfig
version v2.2.20
boardType 6
mdns 1
mqttHost yyy.xxx.dk
mqttPort 1883
mqttClientId ams
mqttPublishTopic ams
mqttPayloadFormat 4
mqttSsl 0
homeAssistantDiscoveryPrefix 
homeAssistantDiscoveryHostname 192.168.1.217
homeAssistantDiscoveryNameTag 
meterBaud 115200
meterParity 8N1
meterInvert 0
meterDistributionSystem 2
meterMainFuse 40
meterProductionCapacity 0
gpioHanPin 16
gpioHanPinPullup 1
gpioApPin 0
gpioLedPinRed 13
gpioLedPinGreen 14
gpioLedRgbInverted 1
gpioVccPin 10
gpioVccOffset 0.00
gpioVccMultiplier 1.000
gpioVccBootLimit 0.0
gpioVccResistorGnd 22
gpioVccResistorVcc 33
ntpEnable 1
ntpDhcp 1
ntpTimezone Europe/Copenhagen
ntpServer pool.ntp.org
entsoeArea 10YDK-1--------W
entsoeCurrency DKK
entsoeMultiplier 1.000
entsoeFixedPrice 2.000
thresholds 5 10 15 20 25 50 75 100 150 65535 3
dayplot 5 1688410800 346304 0 552 353 375 274 214 0 7 3 16 10 69 6 4 39 96 89 722 1032 403 520 444 509 480 513 757218 0 0 0 0 0 86 668 1168 1517 1404 1445 1291 420 647 158 510 279 1 0 0 0 0 0 0
monthplot 6 1688335200 341047 0 6514 5107 5244 3106 3714 5556 5270 5347 4221 4186 3293 4750 6553 4935 6346 4395 4103 5581 6684 5911 6177 6463 4318 2599 2832 6082 5838 6174 5141 4028 5866 747624 1875 10061 14367 15346 14041 13201 13053 14860 13998 15027 14571 15503 14732 12812 7744 10566 14158 12139 5220 7822 4763 11455 9975 13716 9513 5669 14165 14272 2942 11106 13258
energyaccounting 6 7 10.21 33.88 335.86 20.12 43.06 762.71 1 1.09 2 0.56 3 1.03 0 0.00 0 0.00 148.95 355.61

```

ArnieO, 2023-07-04:
We have now had other cases reported on the same issue (reboot loop), and are trying to figure out what is going on.
**Is your power meter (too) a Kamstup Omnia E-meter?**

The `HAN: Footer checksum error` is a known issue on meters that send payloads each second, @gskjold is looking into it so it will surely be fixed within shortly.

gskjold, 2023-07-04:
This is an interesting problem indeed.
Regarding checksum error, I believe this is due to lacking buffer size. I have made this configurable in upcoming patch release.

ArnieO, 2023-07-04:
Unfortunately, there is no way of verifying in the [Kamstrup published documentation](https://www.kamstrup.com/da-dk/elloesninger/intelligente-elmaalere/meters/omnia-e-meter/documents) whether Omnia E-meter complies with the P1 "standard". 

I will contact them and try to obtain such information. 🤞

ArnieO, 2023-07-05:
I received from Kamstrup a "User manual" for Omnia E-meter. 

It turns out that there are two versions of this meter, one (cheaper, I assume) version without 5V on the RJ12 contact:
[image]

I think this is a really strange decision, and I have no idea whether it is implemented in hardware or in the meter firmware. Nowadays it is not uncommon that options are implemented by firmware to avoid having multiple hardware versions. So... only a wild guess from my side of course - but maybe your meter got a firmware update that switched off the 5V. 

Could you please check which version you have, and revert?

syska, 2023-07-05:
This seems like I have the DSMR P1.

[image]

Also important to note that it actually worked with power from the box for the first week. Then it started the reboot loop.

So if the company https://dinel.dk/ (handles the installation of the unit) or https://www.aura.dk/ ( where I buy my electricity ) turned it off with a firmware update.

If you mean if I can revert the firmware update if any, then no ... I'm not in control of it.

But maybe I can contact the company ... 

But if +5v was turned off by a firmware update ... would it then even be able to get into a reboot loop?

ArnieO, 2023-07-05:
Thank you for clarifying and: Sorry - I was not clear regarding my intentions of asking. I just want to get as much information as possible. 
I agree - if 5V was turned off it would be dead without power from USB. 

No need to contact them.
We have an other report regarding reboot looping, @gskjold is looking into the code to see if it could be some "edge situation" there.

syska, 2023-07-05:
@ArnieO 

No problem. Please let me know if you want me to try anything else.

I still LOVE the product ... the very very very fast responses with feedback. That is how all companies should handle there customers ... so issues or not ... I will buy again and recommend the product.

ArnieO, 2023-07-05:
Thank you for your kind words! 😄 

We are still running this without any advertising, focus is instead on tight follow-up of existing customers, and design adjustments of both hardware and firmware. Satisfied customers are great sellers! 

And you are very welcome to register feedback on [the product page](https://amsleser.no/hjem/10-1-pow-p1.html) if you can spare a minute. Scroll down and use this button:
[image]

syska, 2023-07-05:
@ArnieO 

Where is that button on the page? I can't seem to find it ... also tried to disable my ad-blocker ... still can't see it.

All I see is this: 
[image]

ArnieO, 2023-07-06:
> Where is that button on the page?

You need to be logged in to post a review. Use the button at the top of the page:
[image]

ArnieO, 2023-07-06:
**Update on this issue**
***************************
Summary:
- We have two customers with Pow-P1 that (most times) goes into a reboot loop when connected to Kamstrup Omnia E-meter as shown in the video linked here: https://github.com/UtilitechAS/amsreader-firmware/issues/584#issue-1737053091
- The two customers are @syska and a person that contacted us by email (not yet active on Github, we will now encourage him to participate in this thread)
- When the Pow-P1 is powered from USB-C, there is no issue.
- Both have confirmed they have an Omnia E-meter which delivers 5V as per the type number. 
- @syska received a replacement board; no change.
- @syska measured voltages on the board while powered from Omnia E-meter only: 5V and 3.3V were correct.

The video shows a very rapid reboot, indicating that the issue is *probably* (we are not 100% sure) firmware related, but as we have not yet been able to reproduce this, we need help to debug. 

In order to get more information on what is going on we need help to capture serial debug data from the ESP32 module. Such information is currently available on a serial port that can be read by plugging an FTDI module into the row of holes on the PCB:
[image]
The Rx/Tx marking on the PCB is **FTDI pins**. 

If one of you two with this issue have an FDTI module and knows how to do this; please come forward, as this would be the easiest solution. If the module reboots due to a firmware error, the ESP32 sends debug information at 115200 baud. 

If you do not have an FTDI module, it is also possible to adjust the firmware so that debug is sent to the serial port. This implies a rebuild of the code - and @gskjold needs to figure out how to do that from PlatformIO IDE (we know how to do it from Arduino IDE, but this project is too complex for Arduino IDE). So if a new build is needed, we need some time to figure that out.

syska, 2023-07-06:
@ArnieO 

Yes, I have a USB to TTL ... but just to make sure, this should work ... I'm a very "hobbyist" hardware person.
[image]

If, should I then just connect Rx to Tx, and Tx to Rx?

Also, is there any tricks to connect wires to the board without soldering? ( It's not really an issue since I can use the spare board to test on )

syska, 2023-07-06:
@ArnieO 

I had another device as you can see here:
[image]

There was a 3.3 selector pin, but Micro USB, then I just have to use a cable, but that easy.

GND to GND
Tx to Tx
Rx to Rx

Is it just standard serial settings and then Baud Rate to: 115200 ?

ArnieO, 2023-07-06:
This one is very familiar! You can plug it into the holes of the card, no need for the jumper wires.

Yes, standard serial settings and 115200 baud. 

What we want to see is what comes out while it is in that reboot loop, only powered from the Kamstrup.
Can you manage to do this close to your energy meter?

syska, 2023-07-06:
@ArnieO 

```
Calculated: 0f9c8aaf8d1c947346f7b5eb4b826a795280f7a2a1666e6b3d581c4799ee5f9a
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x4004c174
(I) AMS bridge started
(I) Voltage: 3.31V
(D) ESP32 LittleFS
(D)  size: 1507328
(I) -----------------------------------------------
(I) --WiFi configuration--
(I) SSID:                 'myssid'
(I) Psk:                  'password'
(I) Hostname:             'house-powermeter'
(I) mDNS:                 'Yes'
(I)
(I) --MQTT configuration--
(I) Enabled:              Yes
(I) Host:                 'domain.foo.tld'
(I) Port:                 1883
(I) Client ID:            'ams'
(I) Publish topic:        'ams'
(I) Subscribe topic:      ''
(I) Username:             'amsreader'
(I) Password:             'EixOT0sdAFRAuiYJGEPq'
(I) Payload format:       4
(I) SSL:                  No
(I)
(I) --Web configuration--
(I) Security:             2
(I) Username:             'myusername'
(I) Password:             'password'
(I)
(I) --Meter configuration--
(I) Baud:                 115200
(I) Parity:               3
(I) Invert serial:        No
(I) Distribution system:  2
(I) Main fuse:            40
(I) Production Capacity:  0
(I)
(I) --GPIO configuration--
(I) HAN pin:              16
(I) HAN pin pullup        Yes
(I) LED pin:              255
(I) LED inverted:         Yes
(I) LED red pin:          13
(I) LED green pin:        14
(I) LED blue pin:         255
(I) LED inverted:         Yes
(I) AP pin:               0
(I) Temperature pin:      255
(I) Temp analog pin:      255
(I) Vcc pin:              10
(I) Vcc multiplier:       1.000000
(I) GND resistor:         22
(I) Vcc resistor:         33
(I)
(I) --Domoticz configuration--
(I) Enabled:              No
(I)
(I) --NTP configuration--
(I) Enabled:              Yes
(I) Timezone:             Europe/Copenhagen
(I) Server:               pool.ntp.org
(I) DHCP:                 Yes
(I)
(I) --ENTSO-E configuration--
(I) Area:                 10YDK-1--------W
(I) Currency:             DKK
(I) Multiplier:           1.000000
(I) Token:
(I)
(I) Temp sensor count:    0
(I) -----------------------------------------------
(I) Connecting to WiFi network: myssid
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x11c
load:0x4004c000,len:0xa28
load:0x40050000,len:0x26f4
SHA-256 comparison failed:
Calculated: 0f9c8aaf8d1c947346f7b5eb4b826a795280f7a2a1666e6b3d581c4799ee5f9a
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x4004c174
(I) AMS bridge started
(I) Voltage: 3.31V
(D) ESP32 LittleFS
(D)  size: 1507328
(I) -----------------------------------------------
(I) --WiFi configuration--
(I) SSID:                 'myssid'
(I) Psk:                  'password'
(I) Hostname:             'house-powermeter'
(I) mDNS:                 'Yes'
(I)
(I) --MQTT configuration--
(I) Enabled:              Yes
(I) Host:                 'domain.foo.tld'
(I) Port:                 1883
(I) Client ID:            'ams'
(I) Publish topic:        'ams'
(I) Subscribe topic:      ''
(I) Username:             'amsreader'
(I) Password:             'EixOT0sdAFRAuiYJGEPq'
(I) Payload format:       4
(I) SSL:                  No
(I)
(I) --Web configuration--
(I) Security:             2
(I) Username:             'yyy'
(I) Password:             'xxx'
(I)
(I) --Meter configuration--
(I) Baud:                 115200
(I) Parity:               3
(I) Invert serial:        No
(I) Distribution system:  2
(I) Main fuse:            40
(I) Production Capacity:  0
(I)
(I) --GPIO configuration--
(I) HAN pin:              16
(I) HAN pin pullup        Yes
(I) LED pin:              255
(I) LED inverted:         Yes
(I) LED red pin:          13
(I) LED green pin:        14
(I) LED blue pin:         255
(I) LED inverted:         Yes
(I) AP pin:               0
(I) Temperature pin:      255
(I) Temp analog pin:      255
(I) Vcc pin:              10
(I) Vcc multiplier:       1.000000
(I) GND resistor:         22
(I) Vcc resistor:         33
(I)
(I) --Domoticz configuration--
(I) Enabled:              No
(I)
(I) --NTP configuration--
(I) Enabled:              Yes
(I) Timezone:             Europe/Copenhagen
(I) Server:               pool.ntp.org
(I) DHCP:                 Yes
(I)
(I) --ENTSO-E configuration--
(I) Area:                 10YDK-1--------W
(I) Currency:             DKK
(I) Multiplier:           1.000000
(I) Token:
(I)
(I) Temp sensor count:    0
(I) -----------------------------------------------
(I) Connecting to WiFi network: myssid
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x11c
load:0x4004c000,len:0xa28
load:0x40050000,len:0x26f4
SHA-256 comparison failed:
Calculated: 0f9c8aaf8d1c947346f7b5eb4b826a795280f7a2a1666e6b3d581c4799ee5f9a
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x4004c174
```

ArnieO, 2023-07-06:
Thank you very much, this is of great help! 👍 

I think this is a hardware issue; not firmware. 

We see that it crashes when trying to connect to Wifi, and reboots with a "POWERON" reason for the reboot. 
So for some reason it seems to loose power abruptly at that point. It does not report "brownout" (low voltage, but not "off"), which indicates that the voltage has been so low that the ESP32 detects it as "poweron".

Hypothesis:

- When connecting to Wifi, there will be a sharp but short increase in current draw; this is known and normal.
- The Kamstrup Omnia detects this, interprets it as an overcurrent issue - and shuts down the voltage. 

Now let us look at the specification.
According to [the documentation from Kamstrup](https://github.com/UtilitechAS/amsreader-firmware/blob/master/doc/Kamstrup%20OMNIA%C2%AE%20e-meter.pdf), the Omnia E-meter "conforms to the properties defined in DSMR v. 5.0.2 P1 companion standard". [Link to the standard.](https://www.netbeheernederland.nl/_upload/Files/Slimme_meter_15_a727fce1f1.pdf) 

This is what it says about overcurrent protection:
[image]

The standard allows for variable load during operation:
[image]

The way I interpret this: 

- As long as the current draw is lower than 250 mA, the meter shall not shut down the 5V, even if there is a rapid spike in current draw. 
- If the current draw passes 300 mA, the current shall be shut off immediately.
- There is a "tolerance zone" between 260 and 300 mA where the power meter **might** shut down the power. If we reach this level (>260 mA), there may be differences between brands and maybe even between individual power meters from one brand (which might be what we're seeing): The Pow-P1 design is known to work without issues on Aidon and Sagemcom power meters. And we have feedback that it works correctly on at least some Omnia E-meters.

We now need to figure out whether the Pow-P1 draws more than 260 mA for a very short period of time when connecting to Wifi. If the margin is too small, necessary design changes must be introduced (we must expect some variation between ESP32 modules). 

I will make a test setup to investigate this, and report the findings back to this thread.

ArnieO, 2023-07-07:
OK, so the measurement results are in - and we must admit the issue is on the design. 

At the point of reboot, when the device first connects to Wifi, it consumes more than 260 mA for around 0.6 milliseconds.
This is more than what is allowed in the specification, that calls for an immediate shutdown if the consumption exceeds 300 mA. 

The short overcurrent pulse is apparently tolerated by the Aidon and Sagemcom meters - but the Kamstrup Omnia E-meter shuts down the power, causing a reboot.

[image]

I will see if there is a suitable fix that could be implemented to prevent this - and get back.

syska, 2023-07-07:
@ArnieO 

WOW, this is crazy ... but I'm happy you found the issue.

But as a Developer myself, not this low level code. I know there is always a reason things break ... 

What could have been the reason it was able to run for about a week before it started the reboot loop? ( I'm pretty sure I didn't change anything in the settings )

ArnieO, 2023-07-07:
> What could have been the reason it was able to run for about a week before it started the reboot loop? ( I'm pretty sure I didn't change anything in the settings )

My best guess is that we are marginally outside of what is accepted by the Omnia E-power design. And in the real world nothing is as simple as in the specifications: Components values will drift a bit during the first part of the life time of an apparatus - and so on. And such drifting could be in either the Pow-P1 or the power meter - or both.

This situation is typically what could happen when a design is not made clearly with to respect limits of the applicable standard or specification - which I must confess is the case here. 
We were simply not aware of these limits when this design was done. 

But there are ways to cope with this, presumably we need to introduce a supercapacitor as charge buffer (as we do on Pow-K and Pow-U).

ArnieO, 2023-07-07:
Better plot (improved grounding of the test setup). Maximum current level is 440 mA.
[image]

OZ1CT-Ben, 2023-07-11:
> so there is no way to feed the ESP32 whit power from pin1 include a
> super-cap...it will stop working after some time ???
>
> The issue here is the short but strong current pulses, which the Omnia
> E-meter do not allow.
>
> As for any design that marginally works, it could stop working after some
> time due to (normal) component drift during first part of their life time.
>
> By introducing a supercap it is possible to pull current more slowly from
> the source (RJ12 pin 1), while being able to deliver high current pulses to
> the ESP32. The average current draw of this device is less than 100 mA.
>
> I'm currently working on finding a good implementation, both for an
> updated design and as a fix for the current boards.

Ok...I would like to use it in the summerhouse and ther is no easy way to
get 230v to any small +5v to feed the esp33...so it would be great to see a
way to fix it.... Maybe a ESP whiteout a USB interface ???

Send from my mobile office.

ArnieO, 2023-07-11:
> Ok...I would like to use it in the summerhouse and ther is no easy way to get 230v to any small +5v to feed the esp33...so it would be great to see a way to fix it....

Yes, I fully understand it is important that it runs correctly from only the power delivered by the meter. 
We have no report on issues on other P1 meters than the Kamstrup Omnia E-meter, which is (as we understand it) being rolled out in Denmark now. 

So the Aidon, Sagemcom and Landis & Gyr are more tolerant, while Omnia E-meter follows the specification more closely, and thus revealed this design marginality. 

> Maybe a ESP whiteout a USB interface ???

ESP32-S2 has built-in USB data interface, there is no extra circuitry to handled that - if that was what you were thinking. The issue is purely providing stable power to the ESP32 while respecting the maximum current draws from the P1 port.

OZ1CT-Ben, 2023-07-11:
If it send TX every 10 sec and the supercap at 1F can't keep the power
up...maybe a idea to only TX every 60 sec(or more) if you use the Kamstrup
as power supply ???
And I was just thinking I'd the USB port did use som power that could be
savede..

Send from my mobile office.

ArnieO, 2023-07-11:
> If it send TX every 10 sec and the supercap at 1F can't keep the power up...maybe a idea to only TX every 60 sec(or more) if you use the Kamstrup as power supply ???

No, that would not help - because the problem is not average consumption but the short and heavy current spikes that the ESP32 pull whenever it uses Wifi.

cathper, 2023-07-12:
I do notice the note on https://amsleser.no/hjem/10-1-pow-p1.html#/1-pow_p1_options-pow_p1_with_enclosure_and_rj_12_cable

> Note (7th July -23): We are seeing issues on some Kamstrup Omnia e-meters, please do not order if you have an Omnia E-meter. The design needs an adjustment.

Because of that note, I was about to search for some other solution than Pow-P1, but then I stumbled upon this github issue, and now I definitely want the Pow-P1 due to the openness and transparency.

Thus, do you envision the fix to current boards to be somewhat simple to implement? Then I'd consider to buy it anyway before a Q3-Q4 update 🙂

ArnieO, 2023-07-12:
Thank you for the kind words! Yes, I prefer to be open about this unfortunate issue. 

Depends a lot what you mean by "somewhat simple to implement"! 😄 
I will go for a solution with a supercap, the same component we use on Pow-K and Pow-U (that run on power meters that deliver significantly less current). 

A low-ESR (ESR= Equivalent Series Resistance) supercap is needed, the types used for clock backup etc are IMHO not up to it for this application.

Initial measurements indicate we get significantly closer to the spec by soldering in such a cap on the 3,3V side on the board; there are free holes on the board that can be used. I am however not entirely satisfied to be "significantly closer to the spec", so a current limitation (a small resistor) might be needed to limit the current draw from 5V - and this is a bit more tricky to implement on the existing design.

Component availability is also an issue; we do not have stock of those capacitors. Some can be salvaged/from failing boards that are scrapped - but their leads are cut short, so... well, we'll see when I get to the "3D" part of the solution.

Stay tuned; work in progress; more information comes within 1-2 days.

ArnieO, 2023-07-12:
**Update**
****

Before proceeding with more testing, we updated the test setup, and made a micro-USB breakout cable for the [Nordic Semiconductor Power Profiler II](https://www.nordicsemi.com/Products/Development-hardware/Power-Profiler-Kit-2), which is an excellent tool for measurements like this. 
[image]
[image]

The reference before adding the needed components was now this, showing the current pulse drawn by the board shortly after each reboot. 
[image]

This is where the power meter says "No!!" and shuts down the current - causing a reboot loop.

The 5V voltage from the power meter is regulated down to 3,3V, and that current regulator has some (but not much) current limitation per design. So let us see what happens if we add a large supercapacitor on the 3,3V rail - which is the easiest to implement:
[image]

This attenuates the current peak after reboot significantly (from around 450 to 350 mA), but we are still outside the specification of 250 mA. 

In addition, we create a new problem: A long inrush current draw if we connect this to the power meter when the supercap is not charged. The specification allows for a "normal duration" sharp inrush current, but not like this, where we pull > 250 mA for almost 1,5 seconds:
[image]

We can also see that there is an initial short peak at >800 mA, which is simply too much. 

So the current flowing into the supercap must be limited. One relatively simple way to achieve this is by adding a resistor in series with the input pin of the voltage regulator. Doing a worst case calculation, we choose:
R = 5.5V / 250 mA = **22 ohm**

Now, the next problem is to find a way to install it (and @cathper even wants it to be "somewhat simple to implement"...). On the current layout, this is not straightforward. The only option I found was:

- Lift pin 1 of U1 (the voltage regulator)
- Insert a chip resistor towards the neighbor component C4
[image]

I tried first to heat up the pad of pin 1 while (supposedly) gently prying the pin upwards, but... 😞 ...I broke it :
[image]

So then... repair and replace, bending pin 1 on the new component before installing it:
[image]

Then... back to the test bench. 
(Before each test the supercap is discharged "manually" using a 22 ohm resistor.)

There is still an inrush current peak at around 360 mA, caused by ceramic capacitors on the 5V rail. This is normal, and is accepted by the specification:
[image]

But look at the current draw while the supercap charges; this is compliant with the specification - with good margin (we could have used a smaller resistor!). After the short inrush peak, the charging of the supercap starts at around 120 mA:
[image]

And what about the peak after each reboot, when the ESP32 reconnects to wifi?
Before the modification it was at around 450 mA, causing the reboot problem.
It is gone! The supercap takes care of the current pulse completely:
[image]

So @cathper : I'm not sure if that was what you call "somewhat simple to implement"! 😅 

**CONCLUSION**
My conclusion is sadly that we will not sell the current design Pow-P1 to customers that will use it on Kamstrup Omnia E-meter. Even if it is doable (as demonstrated), it is just too much work to do this modification manually. The chip resistor that was used is a size 0805, meaning it is around 2mm long. So this is precision work and the access between two tall pushbuttons is not easy.

This solution will be implemented in Pow-P1 layout v1.2 that we expect to start selling hopefully during Q3 (or worst case Q4).

**SOLUTION FOR CUSTOMERS THAT ARE STUCK**
As far as I know, we have two customers now that have this reboot problem on Omnia E-meters. I will implement this modification for them, and send them new boards. We know who both of you are, and will try to get an upgrade done on a board #2 during this week or weekend at latest; shipping at latest early next week. Feel free to contact me at post@amsleser.no if you want!

Puh... case closed! 😅

closed 2023-07-12 by ArnieO

cathper, 2023-07-12:
Nice! Perhaps a stupid question: Would it be possible to leave U1 and instead pop C4 and remount it with the 22 ohm resistor "partly underneath" (just the one leg of) C4?

It is most certainly beyond my soldering skills to do that fix, so I'll ... "lubricate myself with patience" and wait for the Pow-P1-v1.2. Will you update this post when it's available to conveniently be notified, or should we subscribe by other means?

ArnieO, 2023-07-13:
> Nice! Perhaps a stupid question: Would it be possible to leave U1 and instead pop C4 and remount it with the 22 ohm resistor "partly underneath" (just the one leg of) C4?

No - the track from pin 1 pad goes to a via hole that connects to 5V plane on the back side, so that will not work. 
But cutting that track is actually an option to lifting pin 1 on U1. Then maybe solder the R-chip "vertically" (on its side). Cutting the track can be done with an 1mm machining bit on a drill, and a steady hand. Thank you for guiding me into that alternative! 😄 

[image]

I think Lectronz now has an option to pre-register for new products that are not in sale yet. We could then just make a new variety of Pow-P1 "that also works on Kamstrup Omnia E-meter". 
I'll investigate that and report back here.

ArnieO, 2023-07-13:
**UPDATE**

We fine tuned the resistor value to reduce its power consumption. 22 ohm would be a bit too much for a size 0805 chip resistor, close to 200 mW.

12 ohms seems to be a good choice, its worst case power consumption (device in AP mode) will be around 100 mW - which is fine. And it still gives good margin to power consumption specifications.

syska, 2023-07-13:
> I do notice the note on https://amsleser.no/hjem/10-1-pow-p1.html#/1-pow_p1_options-pow_p1_with_enclosure_and_rj_12_cable
>
>
> Because of that note, I was about to search for some other solution than [...]

Yes, it's been really amazing.

Awesome support from day 1 when I started seeing the issue.

I also got to use some of my different electrical components I had in storage to help with the debugging  :-)

> 
> Thus, do you envision the fix to current boards to be somewhat simple to implement? Then I'd consider to buy it anyway before a Q3-Q4 update 🙂

It's amazing being able to follow the live usage of power.

Now I just need a solution for district heating ... but it does not seem like the Kamstrup unit have the same options and Home Assistant does not have support for this data yet.

ArnieO, 2023-10-31:
An updated Pow-P1 which will run alså on Omnia E-meter is in production now, will be ready for sale early December.

cathper, 2024-01-20:
Although not completely relevant for this issue it at least relates.

I notice that https://www.amsleser.no/hjem/10-1-pow-p1.html#/1-pow_p1_options-pow_p1_with_enclosure_and_rj_12_cable still says:

> Note (7th July -23): We are seeing issues on some Kamstrup Omnia e-meters, please do not order if you have an Omnia E-meter. The design needs an adjustment.

So I'm wondering if the product page just need an update, or if the new version is still not available to buy?

Thanks. 🙇‍♂️

gskjold, 2024-01-21:
I believe the new version is available.@ArnieO ?

ArnieO, 2024-01-21:
This is one of the "curses" of having two sales points... We have updated the description on https://lectronz.com/products/pow-p1-reader-for-power-meter-with-p1-port:
[image]

...but had forgotten to update on amsleser.no.

@cathper : Will update description on amsleser.no now, thank you for the kick on the leg! 👍 

And in fact, we are now shipping only the new version - so I will modify the text on Lectronz too.