UtilitechAS/amsreader-firmware #585: No HAN data received

sm4upt, 2023-06-04:
Since few days ago i don't receive HAN data any more , HAN is red and message "HAN: No HAN data received for at last 30s"
So my question is how to continue fault finding? If its meter or reader that causes this error?

**Hardware information:**
 - Country: Sweden
 - Meter: Kamstrup
 - Encryption enabled No
 - AMS reader: Pow K+ ESP32s2
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.19
 - MQTT: yes
 - MQTT Home assistant
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: 2400 8N1
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]
[image]

ArnieO, 2023-06-04:
Could it be that your grid company has updated the meter to send according to HAN-P1 protocol? I believe some grid companies in Sweden are upgrading the firmware in the meters, and our current version of Pow-K can only read HAN-NVE protocol. 

Datagrams from the meter are indicated by blue flashing of the LED; do you see such flashing?

sm4upt, 2023-06-04:
LED is flashing red

ArnieO, 2023-06-04:
OK, only red flashing indicates the same as the red HAN: Missing data. 
I advice you to contact your grid company and check the status of your HAN port. 
Did they implement changes recently?

sm4upt, 2023-06-08:
I get answer from my grid company yesterday and they don't have made any changes, but their suggestion was to restart the meter, they have experience about power supply issues from Kamstrup to HAN readers without external power supply so i did restart yesterday and it's has been working since that,

closed 2023-06-11 by gskjold

jens0804, 2023-08-08:
how to restart the meter?

ArnieO, 2023-08-08:
> how to restart the meter?

1. Some meters can be remote rebooted by the grid company. If so, this could be the easiest option.
2. In most installations, the main fuse/breaker is on the grid side of the meter. If that is your case: Open the main fuse/breaker, wait 30 seconds, close the main fuse/breaker again.

TheYves, 2023-12-15:
I had the exact same issue. My grid company said they can't reboot the smartmeter, so I had to disconnect the house from the grid. It actually solved the issue, but I hope that this doesn't happen regularly. :-(

ArnieO, 2023-12-18:
> I had the exact same issue. My grid company said they can't reboot the smartmeter, so I had to disconnect the house from the grid. It actually solved the issue, but I hope that this doesn't happen regularly. :-(

Which country is this? 
This is a strange issue. Several persons who had this problem got a remote reboot (of some kind) done by the grid company. I have emailed one grid company in Denmark that says they have no such option, all they could do was to deactivate and reactivate the HAN port (but maybe that does the trick?).

So I emailed a contact at Kamstrup this morning to find out what could be the reason that different grid companies handle such requests so differently. Not all installations have the main fuse/breaker on the street side of the meter, so it would be great to come to the bottom of this.

TheYves, 2023-12-19:
@ArnieO thanks for looking into this. I'm located in Switzerland and my provider is CKW.

I'm looking forward to hearing what your contact person has to say. Wouldn't it make sense for Kamstrup to investigate the actual problem on the meter? In my opinion, it doesn't really matter whether the grid company can restart or not, it shouldn't be necessary anyway.

ArnieO, 2023-12-19:
> @ArnieO thanks for looking into this. I'm located in Switzerland and my provider is CKW.
>
> I'm looking forward to hearing what your contact person has to say. Wouldn't it make sense for Kamstrup to [...]

We just had a similar situation in Denmark, where the grid company also said they couldn't do a remote reboot. Strange thing is - all (I think) grid companies have a software tool (VisionAir) that they use for managing the meters. Some grid companies willingly do "something" - and the HAN port then works again. Others say they cannot do it. 

The first response from the Kamstrup contact gave no clear answer. He could not say without knowing why the meter stopped pushing data, and indicated Kamstrup could be interested in receiving such a meter back for investigation.

We have seen a number of these cases now; maybe around ten in total over the close to 1500 Pow-K units currently operational in Norway, Denmark, Sweden, Switzerland. So I gave him more information on what is typically happening. 

The problem Kamstrup could have with investigating this could be that after the meter is disconnected and reconnected, it works as it should - at least for a while.

ArnieO, 2023-12-19:
The feedback we get from Kamstrup is that the grid company should contact Kamstrup based in the System Support Agreement they have - and ask for guidance on how to reset/reboot meters that behave in this way. 

We at amsleser.no do not have such a Support Agreement, so we would not succeed if we tried to contact the relevant department at Kamstrup - which is fully understandable.

TheYves, 2023-12-19:
I can imagine how hard it is to debug this issue. In my case, the issue appeared one year and 5 days after installation of the Pow-K+. So it might be some kind of (buffer) overflow or something like this.

If I understand you correctly, the smart meter is pushing the data and the module doesn't have to activate or request this somehow? If this is the case, then the issue has to be on the smart meter and not on the HAN module.

Let me know when you heard back from your contact and you need any additional information from me.

ArnieO, 2023-12-19:
> I can imagine how hard it is to debug this issue. In my case, the issue appeared one year and 5 days after installation of the Pow-K+. So it might be some kind of (buffer) overflow or something like this.

Very unlikely. We have no indication supporting that hypothesis. 

> If I understand you correctly, the smart meter is pushing the data and the module doesn't have to activate or request this somehow?

Correct, the meter pushes serial data each 10 second. No activation or request needed, there is no communication from the reader to the meter. 

It is known that the meter can/will shut down power to the HAN port if it consumes too much - as the more measurement-sensitive parts of the meter hav priority. If HAN port power is shut down in that way, it will be restored after some minutes, and the data pushing shall then also restart.

We do not know if the cause of these (very infrequent) "HAN port stop" problems have been preceded by a power shutdown to the HAN port. 

> Let me know when you heard back from your contact and you need any additional information from me.

This was the short summary/conclusion of the dialogue (that I aimed at understanding how the meters can be remote reset): https://github.com/UtilitechAS/amsreader-firmware/issues/585#issuecomment-1862700832

I have now contacted a grid company in Denmark that had a customer in this situation, and requested them to contact Kamstrup based in their System Service Agreement - and ask for guidance on how to remote reset a meter. My aim is to be able to guide customers that experience this on how to approach their grid company.

ArnieO, 2024-02-29:
We finally got to the bottom of this, and issued a blog post today with information on how the Kamstrup Omnipower issue ("No HAN data") can be remedied. 
https://www.amsleser.no/module/ets_blog/blog?id_post=33