UtilitechAS/amsreader-firmware #587: Problems with Accumulated cost calculation

ArnieO, 2023-06-06:
In case anyone else sees the same; this is now a known issue:

A customer has reported an issue with the accumulated cost estimate:
[image]

In this case it is easy to spot. 
The estimate on my own device looks reasonable for the same period, but that calculation cannot be trusted.

rogerthn2019, 2023-06-07:
Not a major issue but checking this it looks like my fixed price is used for both Import and Export. I do have a fixed price on import and spot + 10 öre on export. Monthly average spot makes it hard for AMS reader to calculate ;-)

gskjold, 2023-06-11:
Fixed in next patch. As for fixed price on export, created issue #590

closed 2023-06-11 by gskjold