UtilitechAS/amsreader-firmware #588: Possible bug: Same energy use every day

christeredvartsen, 2023-06-09:
**Describe the bug**
In the dashboard the reader reports that I use the same amount of energy basically every day, which I find somewhat dubious. See screenshot below. The hourly usage seems correct though.

**To Reproduce**
Not sure if/how others can reproduce this.

**Expected behaviour**
Different numbers I guess...

**Screenshots**
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Aidon, model 6534
 - Encryption enabled: no
 - AMS reader: Pow-U+
 - M-bus adapter: ?

**Relevant firmware information:**
 - Version: v2.2.19 (updated today from v2.2.18)
 - MQTT: no
 - MQTT payload type: N/A
 - HAN GPIO: ?
 - HAN baud and parity: ?
 - Temperature sensors: [e.g. 3xDS18B20]
 - ENTSO-E API enabled: no

**Additional context**
Add any other context about the problem here.

ArnieO, 2023-06-09:
Huh... interesting. Never seen that before, looks strange indeed. 
And if you sum up the last 24 hours (dayplot), the sum seems to be around 55 kWh.

Can you add "/monthplot" to your ULR and show us the numbers behind the bars in the monthplot?
An while you're at it: Can you also download a config file and share with us (info page / Backup & restore / Click Download button). Remove any sensitive content you may have before sharing.

I assume it was the same before you upgraded to 2.2.19? 
You are one of very few that has run v2.2.18 for a while, as it had a reboot error - but that should not have impacted the calculations.

tronde-ams, 2023-06-09:
You mean /monthplot.json (with .json) I think?

Does it exist a list of these "secret" lists we can pull from the adapter?

ArnieO, 2023-06-09:
> You mean /monthplot.json (with .json) I think?
> 
> Does it exist a list of these "secret" lists we can pull from the adapter?

Yes - sorry; you're right. /monthplot.json 

The Wiki here is not up to date, but I think I documented this in the Firmware User Manual (downloadable from the product pages of https://amsleser.no --> Vedlegg).

gskjold, 2023-06-11:
Initially this looks to me like it has created an average because it haven't had data for monthplot in a while, although 73 on the last day is kind of weird. Only thing I can think of is that you have been running 2.2.16 (which had an data collection bug), this would produce this average first midnight after upgrade. Is it still wrong?

christeredvartsen, 2023-06-16:
Checked `/monthlot.json` now:

```json
{
"i01": 62.1,
"i02": 62.1,
"i03": 62.1,
"i04": 62.1,
"i05": 62.1,
"i06": 62.1,
"i07": 62.1,
"i08": 62.1,
"i09": 62.1,
"i10": 70.68,
"i11": 64.04,
"i12": 57.2,
"i13": 51.07,
"i14": 53.92,
"i15": 49.99,
"i16": 76.23,
"i17": 76.23,
"i18": 76.23,
"i19": 76.23,
"i20": 76.23,
"i21": 76.23,
"i22": 76.23,
"i23": 76.23,
"i24": 76.23,
"i25": 76.23,
"i26": 76.23,
"i27": 76.23,
"i28": 76.23,
"i29": 76.23,
"i30": 76.23,
"i31": 62.1,
"e01": 0,
"e02": 0,
"e03": 0,
"e04": 0,
"e05": 0,
"e06": 0,
"e07": 0,
"e08": 0,
"e09": 0,
"e10": 0,
"e11": 0,
"e12": 0,
"e13": 0,
"e14": 0,
"e15": 0,
"e16": 0,
"e17": 0,
"e18": 0,
"e19": 0,
"e20": 0,
"e21": 0,
"e22": 0,
"e23": 0,
"e24": 0,
"e25": 0,
"e26": 0,
"e27": 0,
"e28": 0,
"e29": 0,
"e30": 0,
"e31": 0
}
```

Here is what the graph currently looks like:

[image]

And here is the contents of the configuration file:

```
amsconfig
version v2.2.19
boardType 7
mdns 1
meterBaud 2400
meterParity 8E1
meterInvert 0
meterDistributionSystem 2
meterMainFuse 40
meterProductionCapacity 0
gpioHanPin 16
gpioHanPinPullup 1
gpioApPin 0
gpioLedPin 2
gpioLedInverted 1
gpioLedPinRed 13
gpioLedPinGreen 14
gpioLedRgbInverted 1
gpioVccPin 10
gpioVccOffset 0.00
gpioVccMultiplier 1.000
gpioVccBootLimit 0.0
gpioVccResistorGnd 22
gpioVccResistorVcc 33
ntpEnable 1
ntpDhcp 1
ntpTimezone Europe/Oslo
ntpServer pool.ntp.org
entsoeArea 
entsoeCurrency NOK
entsoeMultiplier 1.000
entsoeFixedPrice 0.000
thresholds 5 10 15 20 25 50 75 100 150 65535 3
dayplot 5 1686895210 81064976 0 1312 1336 1304 1248 1360 4096 3760 3696 3808 2784 1736 1736 1440 1416 1960 2192 1552 1592 2184 1480 2072 1808 1328 1360
monthplot 6 1686866410 81051632 1 62100 62100 62100 62100 62100 62100 62100 62100 62100 70680 64040 57200 51070 53920 49990 76230 76230 76230 76230 76230 76230 76230 76230 76230 76230 76230 76230 76230 76230 76230 62100
energyaccounting 6 6 0.00 0.00 0.00 0.00 0.00 0.00 1 4.33 2 4.41 3 4.33 4 4.87 10 5.00 0.00 0.00
```

> I assume it was the same before you upgraded to 2.2.19?

Yes, the graphs prompted me to upgrade, hoping the issue would go away. The numbers do look more correct after the upgrade.

gskjold, 2023-06-16:
Thanks, I'll write off this issue as an aftereffect from bug in  2.2.16. Please reopen if the issue returns

closed 2023-06-16 by gskjold

tfriberg, 2024-02-24:
In the AMS reader UI I get a static 35 kWh energy use per day. In HA it shows the correct values. Is it a/the same bug? I am running v2.2.28 and the Pow-P1 reader.

[image]

Corresponding graph in HA. Historic data missing as I haven't managed to re-map historic values from my Tibber Pulse to the Pow-P1 after switching readers.
[image]

tfriberg, 2024-02-28:
Seems it has started working after a couple of days. Ignore for now.
[image]