UtilitechAS/amsreader-firmware #590: Separate fixed price for import and export

gskjold, 2023-06-11:
Fixed price is currently used for both import and export. Should be separated and independently optional

MySmartHouseDk, 2023-06-14:
This is very much needed as the settings as it now is useless if you're both importing and exporting.

wknutsen, 2023-08-28:
This is also a problem for the multiplier setting. In Norway we only have vat on import.
It would also been nice to have the possibility to add the fixed extra cost on top of spot price all power companies is adding.

To make things even more complicated when exporting there is also a fixed reduction pr. kWh you have exported on the grid bill.

bardahlm, 2023-11-24:
From 2024-01-01 the Norwegian grid companies will change to net import or export per hour. So if you export 1 kWh and import 2 kWh the same hour you will pay for the net, ie 1 kWh. In contrast to the current system where you sell 1 kWh and buy 2 kWh.

gskjold, 2023-11-24:
I thought all norwegian meters always reported net import/export? Are you saying that your meter show both import nad export at the samme time?

bardahlm, 2023-11-24:
> Are you saying that your meter show both import nad export at the samme time?

Yes.

These is my counters for four hours today, note that both counters increase every hour:

|data_tPI|data_tPO|
|---------|---------|
|134417.42|11482.4|
|134418.22|11483.7|
|134419.79|11484.69|
|134420.71|11486.66|

gskjold, 2023-11-25:
Of course, I was thinking instant power... Forgot the power company actually charges by the hour :laughing:

bardahlm, 2023-11-25:
You're correct on instant power. It's either in or out. 
[image]

gskjold, 2023-12-25:
See #252 for a build with proposed solution for price configuration that separates import and export prices

closed 2023-12-25 by gskjold