UtilitechAS/amsreader-firmware #595: 2.20.0 very unstable

Sommerdal-ufm, 2023-06-14:
**Describe the bug**
A clear description of what the bug is.
Web interface very slow.
When answering it shows that it is recently restarted and current hour cost is 0.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: [e.g. Norway]
 - Meter: [e.g. Aidon]
 - Encryption enabled [yes/no]
 - AMS reader: [e.g. Pow-U, ESP32 etc]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: [e.g. 2.1.0]
 - MQTT: [yes/no]
 - MQTT payload type: [e.g. JSON]
 - HAN GPIO: [e.g. GPIO5]
 - HAN baud and parity: [e.g. 2400 8E1]
 - Temperature sensors [e.g. 3xDS18B20]
 - ENTSO-E API enabled: [yes/no]

**Additional context**
Add any other context about the problem here.

ArnieO, 2023-06-14:
Strange issue, and nobody else has reported this.

- I understand this is a Pow-K+ that you have had running since around start of May - is that right?
- You are sure this started when you updated to 2.2.20? 
- No other changes in your Wifi network at or around the same time?
- Do you see any error blinks on the device (ref Firmware User Manual chapter 10)?
- Can you show us a screenshot of your config page?

Sommerdal-ufm, 2023-06-15:
Pow-k - start of May - Correct

I never had this issue before 2.2.20.

It is only with 2.2.20 I've seen an uptime reset with no reason.

No change to network.

Haven't looked at device during problem.

I'll attach a screenshot in a while.

ArnieO, 2023-06-15:
It could be there is an issue with the card for some reason (in which case we'll send you a replacement). We have had very few suck issues - but it cannot be ruled out.

But if it started with the firmware update it would be a strange coincidence if it is HW related. And we have many users on v2.2.20 running it without issues. So... looking forward to your screenshot!

What is the reboot reason it states on the Info-page?
[image]

Sommerdal-ufm, 2023-06-15:
I dont think that the problem is HW related.

I've attached the required screenshots.
[image]
[image]

gskjold, 2023-06-16:
There isn't really much code change in 2.2.20 compared to 2.2.19, it's mostly just user interface which does not affect firmware stability. Reboot with reason Vbat power on reset sounds like there was a power cycle on the chip to me.

Sommerdal-ufm, 2023-06-16:
That is also what I thought.
But I had it twice in 2-3 days, so I assumed it could be a bug.
Feel free to close.

ArnieO, 2023-06-16:
We have seen some boards that start rebooting by themselves like that - with a shorter or longer period. 

I have long-term tested identical boards with identical firmware and identical setup in my Kamstrup meter, and have concluded that there is a HW problem on some ESP modules. The board I have been using for almost a year now reboots a few times per day like that, but it is not often enough to be a real problem for me. 
That is what you see here, 1 hour uptime and same reboot as your board: https://github.com/UtilitechAS/amsreader-firmware/issues/595#issuecomment-1593081526
The problem has not become worse since I started using the board permanently in August last year. 

If this is a problem to you, please email us at post@amsleser.no to get a replacement board.

Sommerdal-ufm, 2023-06-16:
> (comment)
> If this is a problem to you, please email us at ***@***.*** to get a

Thanks. But I really dont consider it a problem yet.
If the problem increases I'll return. 👍

closed 2023-07-27 by ArnieO