UtilitechAS/amsreader-firmware #596: Loosing stored data

leakim1957, 2023-06-16:
I have v2.2.20 and last night at 01.00 all stored data both "Energy use last 24 hours" as well as "Energy use last 31 days" was lost, see picture. This is not the first time this happends. I remember at least two other occations after the HW was replaced in january.
[image]

gskjold, 2023-06-16:
Sounds like there could be some sort of edge case where this happens. Would have to look into that. Does your device have access to internet?

leakim1957, 2023-06-16:
Hi Gunnar
Yes, internet access 24/7.
If there should be an internet drop out at some time, why would that erase the memory stored?

///Mikael

gskjold, 2023-06-19:
It shouldn't, but if it does not have internet access, it gets time from the meter which in some cases are not completely reliable, and if the time drift is too high, the data is cleared

leakim1957, 2023-06-19:
Ok, I will check the connection.
Can you specify the time drift,  Is it seconds, minutes or hours?

gskjold, 2023-08-18:
I've tried to find some way to reproduce this, but I have yet to succeed. Could you give me some more information:

Country:
Manufacturer and model of meter:
AMS reader hardware:

leakim1957, 2023-08-18:
Hi Gunnar
See my answers below in red.
I would like to add that I have only had data saved for a maximum of 5 days in a row for the past few months.
Sometimes only 1 day, sometimes 2-3 days.
///Mikael BW

Country: Sweden

Manufacturer and model of meter: Sagemcom T211

AMS reader hardware:

-

ArnieO, 2023-08-18:
@leakim1957 That did not work. You need to log into Github and write your message there.

leakim1957, 2023-08-18:
Hope this will be better:-)

Country: Sweden

Manufacturer and model of meter: Sagemcom T211, see picture
[image]

AMS reader hardware: See picture
[image]

ArnieO, 2023-08-18:
@leakim1957 You closed the issue - was that on purpose, or was it a mistake?

OK - you are running a Pow-P1 on a Sagemcom in Sweden. 
We have several such installations running, I don't think the hardware is an issue. 

Sounds more like some strange edge issue, probably related to other reports lately of missing/lost data. 
@gskjold is working on it, trying to recreate the issue.

gskjold, 2023-08-18:
Got two issues regarding loosing data, both Sagemcom, do you know of any more cases @ArnieO  ?

ArnieO, 2023-08-18:
> Got two issues regarding loosing data, both Sagemcom, do you know of any more cases @ArnieO ?

None other than those reported in issues here on Github.

leakim1957, 2023-08-18:
It was a mistake, I am not used to work in github

stigvi, 2023-08-27:
Some kind of issues with mine too. Uptime is 34 days, but only the last 19 have any data.

[image]

gskjold, 2023-08-27:
Interesting indeed, something definitely happened here

stigvi, 2023-08-28:
> Interesting indeed, something definitely happened here

Hi. I see from my Tibber App that 8th of August was rather special because my energy cost that day was NOK -1.90. I don't have any entsoe prices downloaded to my unit, so I am not able to see any connection to the missing data. But maybe there is?

gskjold, 2023-08-31:
Good to think outside the box here, thanks for that. I am not able to see a connection there either, but will keep it in mind

gskjold, 2023-09-05:
I finally understand what is going on here now, will fix this for next patch

leakim1957, 2023-09-07:
Can´t wait to test it :-)

senspix, 2023-09-18:
Possibly the same issue but with month plot. Also Sagemcom T211 device in Sweden.
[image]

closed 2023-09-21 by gskjold

leakim1957, 2023-09-29:
Hi. Loading screen shot of system after installing latest version. Let´s see what happends with stored data as time goes by.
///Mikael

[image]

leakim1957, 2023-10-12:
Hi, just when I hoped the problem was solved, it has appeared again. 
No problems since last update but tonight at 0200 it went blank and no update of prices either.
///Mikael
[image]

leakim1957, 2023-11-13:
[image]

31 days in a row without loosing data. Well done!!!
///Mikael

ArnieO, 2023-11-13:
So - let us hope this bug is dead and gone for ever now!