UtilitechAS/amsreader-firmware #598: Energy usage last x days shows current day (with wrong nr of kWh of course)

swidtwee, 2023-06-19:
**Describe the bug**
Energy usage last x days shows the current day as well. This happens not from the start of the day but from 21-22 hours or something it shows eg. 44/54 kWh for current day (while total consumption at that time only is 25-ish kWh). Screenshot from 23:15 on 19th of June already shows the 19th. (After end of day the correct number of kWh's is shown in the chart.)

**Expected behavior**
Current day should not be shown in the chart energy usage last x days.

**Screenshots**
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - AMS reader:Pow-U,

**Relevant firmware information:**
 - Version:2.20.0 (but exists already since a few versions)
 - MQTT: no

gskjold, 2023-06-20:
Day graph is dependent on correct timezone, what timezone is set in your configuration?

ArnieO, 2023-06-20:
> Day graph is dependent on correct timezone, what timezone is set in your configuration?

But can the erroneous value he sees until the day is over be explained by timezone error?

gskjold, 2023-06-20:
Thats true, not unless 20ish kwh was used in the last 2hrs of the day.

ArnieO, 2023-06-20:
Sounds like something in the accumulation calculation on his device is messed up?
It is maybe worth while to try a Factory Reset?
@swidtwee : If you do that and want to preserve the historic values, you can export it first by going to the Info-page and use the "Backup & restore" feature.

tronde-ams, 2023-06-21:
Norway, Aidon 3p IT, generic ESP32.

I did see this a few hours after (in the evening) I updated to 2.2.20. I took a screenshot, but since it disappeared at midnight, and newer came back (as far as I know), I deleted it and considered it to be a random thing. Mine did also show to much, and it was not a combined value of yesterday and today. I will try to catch an eye on the graphs in the evening.

tronde-ams, 2023-06-21:
Some bugs are hiding here.

At 21:00 it was OK.

[image]

`mqttPayloadFormat 2
mqttSsl 0
meterBaud 2400
meterParity 8E1
meterInvert 0
meterDistributionSystem 1
meterMainFuse 35
meterProductionCapacity 0
gpioHanPin 16
gpioHanPinPullup 1
gpioApPin 0
gpioLedPinRed 2
gpioLedPinGreen 15
gpioLedRgbInverted 0
gpioVccOffset 0.00
gpioVccMultiplier 1.000
gpioVccBootLimit 0.0
ntpEnable 1
ntpDhcp 1
ntpTimezone Europe/Oslo
ntpServer pool.ntp.org
entsoeArea 10YNO-1--------2
entsoeCurrency NOK
entsoeMultiplier 1.000
entsoeFixedPrice 0.000
thresholds 5 10 15 20 25 50 75 100 150 65535 3
dayplot 5 1687377612 119591360 0 848 760 840 1400 624 712 1288 1480 1928 1792 1064 1416 1832 2928 1072 1128 1240 1328 880 800 1128 1560 2176 1664
monthplot 6 1687298412 119562160 0 26032 28160 28432 25440 28504 25672 27632 24976 26248 19192 22480 25080 23920 28440 28840 25656 24512 25376 23600 28944 26800 30144 27600 29064 26936 22440 30592 27616 32360 27232 26632
energyaccounting 6 6 26.41 398.35 791.95 0.00 0.00 0.00 1 2.72 7 2.42 21 2.92 13 2.40 5 2.43 986.62 0.00`

tronde-ams, 2023-06-21:
Did not update the erroneous bar at 23:00, but OK after 00:00. 
.
[image]
[image]

gskjold, 2023-06-22:
> Did not update the erroneous bar at 23:00, but OK after 00:00.

Thank you, that was some useful info! I got an idea of what this is now, will investigate :)

tronde-ams, 2023-06-22:
Same again. 

OK at 21:00.
Error at 22:00
Same vale at 23:00
OK at 00:00

.
[image]
[image]
[image]
[image]

closed 2023-07-09 by gskjold