UtilitechAS/amsreader-firmware #599: Logged data lost

Nealux, 2023-06-20:
I have also a case with lost historic data (could be similar to #596)
Data for the last month was lost the day before yesterday.
The device has access to internet 24/7.
No problem seen with the internet at the time.
A brownout was registered as cause of last reboot.
I assume that data for the last 24 H was lost at the same time but as I did not look at the device yesterday I can not be sure.
[image]
 
Country: Sweden
Meter: Sagemcom
Encryption enabled : no
AMS reader: Prototype

Version: 2.2.20
MQTT: no
HAN baud and parity: 115000 8N1
Temperature sensors No
ENTSO-E API enabled: yes

ArnieO, 2023-06-20:
Interesting situation. 
I understand you are running this on an "old" prototype Pow-P1 (you were one of those who helped us during the design phase). 

If so, it means you device has been running for around a year. Have you ever seen brownouts before? (Or before the reboot indication was added to the firmware: unexpected reboots?)

@gskjold : Do you see any possibility for the firmware to loose historic data in case a reboot happens at the worst possible moment?

Nealux, 2023-06-20:
As far as I recall this is the first time I've seen brownout as a reboot reason.
But before reboot reason was added I've had a few "spontaneous" reboots.

gskjold, 2023-07-02:
I'm trying to figure out what this could have been, but I do not have a proper answer yet. I will keep thinking about this and see if I am able to reproduce

ArnieO, 2023-07-27:
@Nealux 
I see now some things in your input that are puzzling:

You are running a prototype version of Pow-P1 which uses a different GPIO for reading data compared to the "production versions". This means that to configure it correctly, you need to add "/vendor" to your URL and "declare" it as a "Generic ESP32-S2", like this:
[image]

After that, you need to go to the config page and select the correct GPIO, which in your case is GPIO18. When selecting "Generic ESP32-S2", you will have a tile there that is not used for the "production" versions. 

Your screendump here above shows that you have selected "Pow-K+" as hardware profile, which reads data on GPIO16. So I am curious to understand how you got this to work at all...??

A brownout on that Pow-P1 prototype design is actually possible (we now know), as despite our efforts the capacitance on the card (that is there to support the ESP while it draws big current spikes while working on Wifi) is insufficient (i.e. the design margin is insufficient). 

If you want to tinker with your board to improve that point, please take inspiration and information from this thread: https://github.com/UtilitechAS/amsreader-firmware/issues/584

Nealux, 2023-07-27:
The reason why it works despite the chosen hardware profile is that the GPIO in the config file has been manually changed to 18.
Is i important to change the hardware profile to Generic or does it only affect the GPIO's in this case?

The brownouts still occurs from time to time but now without any loss of data.
The board was modified with a capacitor (100 uF) across +5 and ground as suggested to cure a problem with the device running as an AP.

gskjold, 2023-08-18:
Got my Power Profiler Kit up and running today. Have tried to reproduce this issue, without luck.
I notice that the power consumption is significantly increased when the wifi is lost and it is searching for the access point (80mA). @ArnieO could this power increase cause the prototype p1 to brown out?

[image]

ArnieO, 2023-08-18:
> @ArnieO could this power increase cause the prototype p1 to brown out?

No, it should not. 
The ESP32 detects brownout at around 2.8V. 
This should be prevented by the MAX811T as explained [here](https://github.com/UtilitechAS/amsreader-firmware/issues/599#issuecomment-1653593897). It resets the ESP if the voltage goes below nominal 3.08V (minimum 3.03V). So the ESP should never see brownout voltage level.

This is why this behavior is so puzzling to me, it looks as if U2 does not work properly. 

@Nealux : Does the device respond correctly to pressing the Reset button? It is connected the Manual reset input of U2.

Nealux, 2023-08-18:
When the Rest button was pushed the device rebooted.
Reason: Vbat power on reset (1/0)

Skickades från E-post<https://go.microsoft.com/fwlink/?LinkId=550986> för Windows

ArnieO, 2023-08-18:
> When the Rest button was pushed the device rebooted. Reason: Vbat power on reset (1/0)

OK, that is correct behavior when the MAX811T disables the ESP module.

Just to be sure which board version you have: Is it one of these?
[image]

Those boards were not really intended for more than short-term testing. Amongst other things, we did not have the USB line protection chip (U6), so the footprint of U6 is strapped with two wires. And as result of the test, we beefed up the board capacitance significantly to ensure voltage was stable while the ESP transmitted. We did not have the right RJ12 sockets, so a cable was soldered in (which is mechanically a quite weak point). 

To be honest, I don't really want to spend more time on trying to remote debug a problem one of those prototypes! 😅

Nealux, 2023-08-18:
Yes I have one of those.
If I understand you correctly you do not want me to report any more problems that I encouter with the device?

Skickades från E-post<https://go.microsoft.com/fwlink/?LinkId=550986> för Windows

[image]<https://user-images.githubusercontent.com/10295178/261644327-64e337e6-062d-49c1-85b1-f20677186770.png>

ArnieO, 2023-08-20:
> If I understand you correctly you do not want me to report any more problems that I encounter with the device?

Ah, sorry - I was not clear. What I meant was that I do not see any benefit in investigating hardware related issues such as the brownout. 

We appreciate all feedback on issues that might be firmware-related. The loss of data seems to be related to firmware - and it is indeed useful for @gskjold to get feedback from anyone experiencing that - in his search for the cause. 
As far as I understand, he has not been able to reproduce it yet.

gskjold, 2023-09-05:
Considering this as duplicate of #596

closed 2023-09-05 by gskjold