UtilitechAS/amsreader-firmware #600: sceurity problem? - "Swap to AP mode if WiFi psk is wrong"

bmork, 2023-06-20:
**Describe the bug**
Possible unauthenticated remote access

**To Reproduce**
- Configure a rogue AP with the SSID used by amsreader, 
- wait for the amsreader to attempt a connection and fail, 
- connect to the amsreader AP and reconfigure it

**Expected behavior**
physical access or authentication required for any amsreader management

**Note**
I've not actually tried this.  It's only based on assumed behaviour from the feature description and code.  But I'd really prefer not to have this feature. At least not by default. Pushing a button for AP mode is not a problem for the rare occasion where the password changed in the wrong order (AP before client).

ArnieO, 2023-06-20:
You're making an interesting point! 
@gskjold : What do you think?

gskjold, 2023-06-21:
Technically possible, good point. Will only default this feature for initial setup

closed 2023-07-02 by gskjold