UtilitechAS/amsreader-firmware #601: C6 Support?

Roarrk, 2023-06-26:
**Describe your problem**
I'm unable to get esp32-c6 working. Specifically espressif esp32-c6-wroom1. Trying to find the correct bootloader/firmware, but it seems like the esp-idf that the tasmota firmware is based on is running on v 4.4 which does not support c6 yet?

gskjold, 2023-07-01:
As far as I can tell this boils down to arduino-esp32 using idf 4.4 in their current releases (https://github.com/espressif/arduino-esp32/releases) , and yes, it does not support c6

gskjold, 2023-07-27:
https://github.com/espressif/esp-idf/issues/10423

Hedda, 2023-08-17:
Requires waiting for Arduino release based on ESP-IDF 5.1 as it supports ESP32-C6, follow -> https://github.com/espressif/arduino-esp32/issues/7852

Jason2866, 2023-08-18:
@gskjold Seems your project is using our fork of Platformio/Arduino.
Don't hesitate to test the experimental build for C6 based on Arduino 3.0/IDF5.1
```
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
platform_packages = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/1511/framework-arduinoespressif32-release_v5.1-90b05eff62.zip
```

gskjold, 2023-08-24:
Thanks @Jason2866 , OneWireNg would not build, so I have stripped dallas temperature sensor from the code for the moment. The build runs, but the linker fails with the following:
```
/home/gunnar/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c6dev/lib773/libWiFiClientSecure.a(WiFiClientSecure.cpp.o): in function `WiFiClientSecure::stop()':
/home/gunnar/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:98: undefined reference to `stop_ssl_socket(sslclient_context*, char const*, char const*, char const*)'
/home/gunnar/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c6dev/lib773/libWiFiClientSecure.a(WiFiClientSecure.cpp.o): in function `WiFiClientSecure::read(unsigned char*, unsigned int)':
/home/gunnar/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:232: undefined reference to `get_ssl_receive(sslclient_context*, unsigned char*, int)'
/home/gunnar/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c6dev/lib773/libWiFiClientSecure.a(WiFiClientSecure.cpp.o): in function `WiFiClientSecure::available()':
/home/gunnar/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:246: undefined reference to `data_to_read(sslclient_context*)'
/home/gunnar/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c6dev/lib773/libWiFiClientSecure.a(WiFiClientSecure.cpp.o): in function `WiFiClientSecure::write(unsigned char const*, unsigned int)':
/home/gunnar/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:202: undefined reference to `send_ssl_data(sslclient_context*, unsigned char const*, unsigned int)'
/home/gunnar/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c6dev/lib773/libWiFiClientSecure.a(WiFiClientSecure.cpp.o): in function `WiFiClientSecure::WiFiClientSecure()':
/home/gunnar/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:37: undefined reference to `ssl_init(sslclient_context*)'
/home/gunnar/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c6dev/lib773/libWiFiClientSecure.a(WiFiClientSecure.cpp.o): in function `WiFiClientSecure::connect(IPAddress, unsigned short, char const*, char const*, char const*, char const*)':
/home/gunnar/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:141: undefined reference to `start_ssl_client(sslclient_context*, IPAddress const&, unsigned long, char const*, int, char const*, bool, char const*, char const*, char const*, char const*, bool, char const**)'
/home/gunnar/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: .pio/build/esp32c6dev/lib773/libWiFiClientSecure.a(WiFiClientSecure.cpp.o): in function `WiFiClientSecure::connect(char const*, unsigned short, char const*, char const*)':
/home/gunnar/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:163: undefined reference to `start_ssl_client(sslclient_context*, IPAddress const&, unsigned long, char const*, int, char const*, bool, char const*, char const*, char const*, char const*, bool, char const**)'
```
Do you have any pointers on how to fix this?

Jason2866, 2023-08-24:
@gskjold We have modified OneWire from Paul Stoffregen to work with C2 and C6.
You find the modified lib here https://github.com/arendst/Tasmota/tree/development/lib/lib_basic/OneWire-Stickbreaker

Jason2866, 2023-08-24:
Are you using the inbuilt https Client/Server? If yes you can't use the fork.
We have removed some Crypto Protocols from mbedtls (real fat...). We do not use mbedtls at all, since it is big and resource hungry. We use BearSSL.

Jason2866, 2023-08-24:
Not really, since we don't use https with Webfrontend. Using the the standard http client from Arduino. For https OTA updates we have written our own https implementation with [Berry](https://github.com/berry-lang/berry) we have included in every Tasmota32 variant.

gskjold, 2023-09-16:
@Jason2866 Is WiFiClientSecure stripped out for good? If so, what is the best way to communicate with a API over HTTPS?

Jason2866, 2023-09-16:
We are using BearSSL

msasdk, 2025-11-27:
Are there any updates on C6 support?