UtilitechAS/amsreader-firmware #604: Kronor in the bar chart but öre on the Y axis, after waiting a while on inactive tab

andreas1974, 2023-07-04:
**Describe the bug**
After leaving AMSReader open on a tab in a Chrome browser, the vertical bars in the "future energy price" returns to show Kr instead of öre. The Y axis still displays Öre, so the bars are hardly visible. Reloading the page restores the correct look. Not sure how long I need to wait until it happens. Working with other stuff, and having a look at the AMS reader every second hour I noticed this three times today.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Amsreader in a Chrome broswer on a PC.
2. Open another tab and wait for an unknown amount of time.
3. Return to the AMS reader tab.
4. See error

**Screenshots**
[image]

**Hardware information:**
 - Country: Sweden
 - Encryption enabled [no?]
 - AMS reader: [Pow-U]
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.20
 - ENTSO-E API enabled: yes

ArnieO, 2023-07-05:
Thank you for reporting, but this is a known issue - will be fixed in upcoming v2.2.21, see https://github.com/UtilitechAS/amsreader-firmware/issues/592

closed 2023-07-05 by ArnieO

andreas1974, 2023-07-05:
I see, I read through all the the **open** issues, before reporting but since it was already closed I didn't notice it! Thank you for dealing with the problem!