UtilitechAS/amsreader-firmware #605: digital input readout

manfi001, 2023-07-05:
Would it be possible to implement readout of digital input(s)? This way it would be possible to monitor for instance a SPD with auxiliary contacts. output to MQTT is sufficient.
Thanks for consideration.

manfi001, 2023-07-05:
SPD is a surge protection device. Nowadays it's mandatory to install these in the distribution system. If it's a model with an auxiliary contact, it could be connected to the reader hardware (a Wemos D1 mini in my case) and the status be monitored without having a look into the distributor box every day

ArnieO, 2023-07-05:
Does such an SPD auxiliary contact have a standardized interface (mechanically and electrically)? Relay output and screw terminals?

ArnieO, 2023-07-05:
Yes, relay output and terminal block. 
[image]

Towards the amsreader: Two wires: One to GND or VCC, the other to a GPIO - with a series resistor. That's it. If the GPIO changes state: Display it on main page + send to MQTT.

My only concern would be the risk of a surge (e.g. lightning strike) propagating to the SPD relay output and into the amsreader device (damaging it), and maybe propagates from there into the HAN port? Hm...

manfi001, 2023-07-05:
exactly!
I don't think your fears of damaging the amsreader are justified

ArnieO, 2023-07-05:
> I don't think your fears of damaging the amsreader are justified
You're most probably right, unless there is a serious lightning strike.

Anyway, I don't foresee that we will do anything more on the HW side than making a GPIO available, as already implemented on next generation. And on the firmware probably some very basic HIGH/LOW reading of that GPIO that can be enabled in the "stock" firmware. Depending on @gskjold priority list for the firmware this could potentially come sooner than the next gen hardware (for use on your Wemos, and on current generation Pow-P1, that already have GPIO available for user).

closed 2023-07-05 by ArnieO