UtilitechAS/amsreader-firmware #609: Smart config

gskjold, 2023-07-11:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_smartconfig.html

henriklund, 2023-07-15:
Looks promising. It would appear that this then requires an app for [iPhone](https://apps.apple.com/us/app/espressif-esptouch/id1071176700) or [Android](https://play.google.com/store/apps/details?id=com.khoazero123.iot_esptouch_demo&hl=en_IN&gl=US)

ArnieO, 2023-07-17:
> Looks promising. It would appear that this then requires an app for [iPhone](https://apps.apple.com/us/app/espressif-esptouch/id1071176700) or [Android](https://play.google.com/store/apps/details?id=com.khoazero123.iot_esptouch_demo&hl=en_IN&gl=US)

Yes, using Smart config will require using the Espressif app.

However, using the app will be optional. 
The current Wifi onboarding method (start Access point, connect to it, open web browser on http://192.168.4.1) will still be available. Those who are used to this method will in fact not notice any change.

bardahlm, 2023-07-20:
Sounds a bit unsecure. The app broadcasts the wifi credentials so it should not be used where those credentials could be misused by rouge users.

gskjold, 2023-07-20:
V2 uses encryption for this process

gskjold, 2023-07-25:
Looking deeper into this, v2 only supports symetric encryptions, meaning it would need a pre-shared key. Any thoughts on this?

closed 2024-01-26 by gskjold