UtilitechAS/amsreader-firmware #612: Time displayed is wrong and all ENSO prices are wrong probably because of wrong date

KennethLavrsen, 2023-07-18:
**Describe the bug**
A clear description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
Upgraded to 2.2.21

Noted all the ENSO prices are wrong. Current price also wrong. 
Then noted that the date in the upper right is one day ahead
It shows 20th July but it is only 19th of July

**Expected behavior**
I expect correct date

**Screenshots**
[image]

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup
 - Encryption enabled yes
 - AMS reader: Pow-K
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.21
 - MQTT: yes
 - MQTT payload type: 
 - HAN GPIO: 
 - HAN baud and parity:
 - Temperature sensors
 - ENTSO-E API enabled: yes

**Additional context**
Add any other context about the problem here.

KennethLavrsen, 2023-07-19:
More information. As I raised the bug report it was past midnight on the 19th Danish time. Danish time with DST is two hours ahead of UTC.

When I downgrade to versions 2.2.18, 2.2.19 or 2.2.20 the time displayed is correct. The prices are wrong and seems to be yesterdays prices.

While being on 2.2.19 time passed 2 AM and suddenly the prices were correct after a reboot.

And upgrading back to 2.2.21 gives the correct date

You seem to not handle the conversion from local time to UTC correctly.   Between midnight and 2 AM (when you are in Central European Summer Time) the prices fetched are from yesterday. And in 2.2.21 you seem to add one to the current date so you show tomorrow.

ArnieO, 2023-07-19:
What is your DNS setting?

ArnieO, 2023-07-19:
OK, that should be fine.
Sorry, I have no idea - but it sounds as if it for some reason got the wrong time synchronization value from internet. 
What are your settings for NTP? Are you using the factory setting pool.ntp.org?

gskjold, 2023-07-19:
I think this is purely a gui issue. Will have a look at it

KennethLavrsen, 2023-07-19:
Time is 3 minutes past midnight on the 20th of July and the GUI shows 21st so error is same as yesterday. Prices seems right though. Also after a reboot

KennethLavrsen, 2023-07-20:
Morning the 20th. Date in UI is now correct again. Prices are still from wrong day. A reboot is required to correct it. This is a serious bug and not just UI. I get wrong prices in Home Assistant with this

KennethLavrsen, 2023-07-20:
Extra info to reproduce and fix.
When time passes midnight, the UI show the date wrong by +1 day, but prices are correct
When you reboot between midnight and 2 AM prices loaded become wrong.

bardahlm, 2023-07-24:
> When you reboot between midnight and 2 AM prices loaded become wrong.

Timezone issue (gmtime() vs localtime())?

In those two hours the date differs between localtime and UTC.

gskjold, 2023-07-25:
Yeah, this is definitely two issues. One being the gui displaying the wrong date. The other that it loads the wrong prices if you reboot between 00:00 and 02:00. But yes, UTC vs CEST. I have a bit of a busy period now, but I'll get around to this one

gskjold, 2023-07-30:
Please check attached firmware containing fixed code:

[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12209431/esp32.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12209432/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12209433/esp32s2.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12209434/esp32solo.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12209435/esp8266.zip)

KennethLavrsen, 2023-07-30:
The zip contains many bin files and not the single file we normally upload.
I have no clue how to program that. Can you do a normal build type bin? I use the s2

gskjold, 2023-07-30:
Inside the zip there should be a firmware.bin that you can upload via gui

KennethLavrsen, 2023-07-30:
Programmed. So far so good. I will stay awake till a little after midnight and check time as well as loaded prices

KennethLavrsen, 2023-07-30:
That code made no difference. At midnight it changed to 1st of August (no reboot needed)  and after a reboot the price is again wrong

gskjold, 2023-07-31:
OK, thanks for the update. I see what I did wrong with the gui. I am not sure about the prices though, are you using your own entso-e api key?

KennethLavrsen, 2023-07-31:
Yes, my own key.  To fix prices I have to reboot outside the time interval 0:00 an 02:00 which clearly shows that it gets the date wrong and fetch the prices from the day before. So it displays a day later and fetch the day before. Something got very wrong when you fixed a date bug some versions ago.

gskjold, 2023-07-31:
OK thanks, I had only adjusted the code when using our intermediate server, because that was obviously wrong. The other code that is used when API key is specified looked OK to me, but obviously it is not. 

And just to make it clear, this is unrelated to the visual bug in GUI. Looking at the code history, this portion have been untouched for quite some time, except for updating the API endpoints hostname in april.

gskjold, 2023-08-18:
@tronde-ams Yes, this is part of the same problem.

Lets try again:
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12376586/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12376587/esp32s2.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12376588/esp32solo.zip)
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12376589/esp32.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12376591/esp8266.zip)

gskjold, 2023-09-21:
Final testing of this feature in issue #641