UtilitechAS/amsreader-firmware #617: Last month consumption calculation

Nealux, 2023-08-01:
I had a look at last month's consumption about one hour ago.
It was 76 kWh.
That is too low.
The correct value is about 315 kWh according to Ellevio.
And that is what I get when I look at the config file too.
What makes this much more confusing is that when I have a look at it about one hour later the value is correct, 315 kWh!
Very strange.
I'm glad I have a screen shoot of the failing calculation.

I have no clue how to reproduce this.

**Expected behavior**
A correct calculation of last month's consumption at all times.

**Screenshots**
[image]

**Hardware information:**
 Country: Sweden
Meter: Sagemcom
Encryption enabled : no
AMS reader: Prototype

Version: 2.2.21
MQTT: no
HAN baud and parity: 115000 8N1
Temperature sensors No
ENTSO-E API enabled: yes

ArnieO, 2023-08-10:
Do you know at which time (approximately) you made the screenshot? 
There is an open issue regarding some inconsistencies shortly after midnight. Maybe this is part of the same issue? #612

Nealux, 2023-08-10:
I don't have the exact time but the screenshot was saved at 14:35 so not around midnight.
Probably around 14:30

ArnieO, 2023-08-10:
> Probably around 14:30

OK, thank you - that is useful to know. 
Did you notice if the device had recently rebooted?

Nealux, 2023-08-10:
I pass on that one as I'm not 100 % sure of how long it had been up and running.
And I only have the screenshot above, one screenshot of the last 31 days consumption and the config file save from that moment.

gskjold, 2023-10-04:
Unable to reproduce

closed 2023-10-04 by gskjold