UtilitechAS/amsreader-firmware #620: MQTT state class total_increasing

rogerthn2019, 2023-08-08:
**Is your feature request related to a problem? Please describe.**
Now and then I do get warnings like
`WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.ams_cf58_dayproduced from integration mqtt has state class total_increasing, but its state is not strictly increasing. Triggered by state 5.23 (5.24)`
I've seen it for dayproduced, monthproduced, dayuse  and monthuse.

**Describe the solution you'd like**
Check new value => previous value before sending mqtt updates

**Describe alternatives you've considered**
Ignore warnings ;-)

**Additional context**
Not a big deal.

gskjold, 2023-08-18:
I guess this would happen if the device reboots. At this point it will not know exactly where it is on the daily counters and does not have any way of knowing. Not sure how to solve that yet

rogerthn2019, 2023-08-18:
Not at reboot for me, `Last boot: 11.07.2023 15:31` and not very often
**grep ams_cf58 home-assistant.log | grep total_increasing | awk '{print $1,$2,$7,$22,$23,$24,$25,$26,$27,$28,$29,$30,$31}'**
`2023-08-07 19:05:11.118 sensor.ams_cf58_dayproduced Triggered by state 5.23 (5.24) with last_updated set to 2023-08-07T17:00:41.719594+00:00.`
`2023-08-07 19:05:11.121 sensor.ams_cf58_monthproduced Triggered by state 130.97 (130.98) with last_updated set to 2023-08-07T17:00:41.719088+00:00.`
`2023-08-07 20:05:11.125 sensor.ams_cf58_dayuse Triggered by state 9.77 (9.78) with last_updated set to 2023-08-07T18:00:41.631956+00:00.`
`2023-08-07 20:05:11.129 sensor.ams_cf58_monthuse Triggered by state 62.9 (62.91) with last_updated set to 2023-08-07T18:00:41.629544+00:00.`

Might be caused by meter register Accumulated active import changing from e.g. 74571493 to 74571490 at hour shift.
I've seen the same for export.

As said, "Not a big deal"