UtilitechAS/amsreader-firmware #622: Realtime export hour is wrong

wknutsen, 2023-08-21:
Export hour kWh is wrong in frontend and in export to MQTT. However realtime day total export and Energy use last 24 hours (kWh) graph seems correct. See "he" on last post what i would expect to be output.

Example from two hour samples from MQTT

{
	"id": "",
	"name": "ANSReader",
	"up": 220063,
	"t": 0,
	"vcc": 3.303,
	"rssi": -73,
	"temp": -127.0,
	"data": {
		"lv": "AIDON_V0001",
		"id": "",
		"type": "6525",
		"P": 0,
		"Q": 0,
		"PO": 305,
		"QO": 470,
		"I1": -0.9,
		"I2": 0.0,
		"I3": -1.4,
		"U1": 239.8,
		"U2": 237.5,
		"U3": 236.7,
		"tPI": 93346.712,
		"tPO": 586.38,  # Accumulated active export
		"tQI": 2307.21,
		"tQO": 8428.96,
		"rtc": 1692619200 = Monday, August 21, 2023 2:00:00 PM GMT+02:00 DST
	},
	"realtime": {
		"h": 0.0,
		"d": 6.1,
		"t": 5,
		"x": 2.97,
		"he": 0.57,  # Export this hour
		"de": 7.2    # Export total today
	}
}

{
	"id": "",
	"name": "ANSReader",
	"up": 223663,
	"t": 0,
	"vcc": 3.31,
	"rssi": -76,
	"temp": -127.0,
	"data": {
		"lv": "AIDON_V0001",
		"id": "",
		"type": "6525",
		"P": 0,
		"Q": 0,
		"PO": 2341,
		"QO": 231,
		"I1": -5.5,
		"I2": 0.0,
		"I3": -5.3,
		"U1": 239.7,
		"U2": 238.1,
		"U3": 238.7,
		"tPI": 93346.712,
		"tPO": 588.07,  # Accumulated active export
		"tQI": 2307.21,
		"tQO": 8429.19,
		"rtc": 1692622800 = Monday, August 21, 2023 3:00:00 PM GMT+02:00 DST
	},
	"realtime": {
		"h": 0.0,
		"d": 6.1,
		"t": 5,
		"x": 2.97,
		**"he": 0.6, # Should have been 588.07 - 586.38 = 1.69 = 1.7 kWh**
		"de": 8.8  # Seems correct
	}
}

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled: No
 - AMS reader: ESP32S2 Pow-U+
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.21
 - MQTT: Yes
 - MQTT payload type: JSON
 - HAN GPIO: Unkown
 - HAN baud and parity: Unknown
 - Temperature sensors: No
 - ENTSO-E API enabled: Yes

gskjold, 2023-08-24:
I can confirm this would be an issue with NVE standard data. Will fix

gskjold, 2023-09-21:
Final testing of this feature in issue #641

closed 2023-09-21 by gskjold