UtilitechAS/amsreader-firmware #625: Correct price calc

PerBuch54, 2023-08-29:
Hi

I would like to se the correct import and export prices.
For export i get the raw price - 0.01 DKK
For import we in DK have a tax of 0.69 DKK pr kWh
And then 25% vat on the total.
W altso have a cable loss fee
It is i 4 differnt timezones during the day
From 0:00
Changes at 6:00
Changed again at 17:00
And then again at 21:00
This is suspended until 1/1-2024
Not needed now and i dont know how a new version would be.

gskjold, 2023-08-31:
This is a larger issue that have been discussed in depth over the last couple of years, but no good solution have presented itself yet. If we can find a good way to cover the grid fee schemes of at least three different countries (with different schemes), we will probably be willing to invest time for a implementation, as this would most likely establish a good base to cover all major grid fee schemes.

In Norway we have over 100 grid operators, giving you (theoretically) over 100 different prices. These fees have some centralized rules, so in practical terms the differences are not that great, but we would still need to find a way to identify what grid fees apply to the user in question.

For more reading, see #252 and #555

PerBuch54, 2023-08-31:
I dont know anything about the prices in Sweden.
My guess are that a seperating the inport and export prices would make sence.
My suggestion are to add a fixed addition and then vat% to both would bee a good start.

PerBuch54, 2023-09-19:
Please tell me how the tarifs are in norway and sweden. I will figure out a way to satisfy all.

ArnieO, 2023-09-19:
This is complex, and I can only answer for Norway.

There are two main sections of the total tariff (what you actually pay for your electricity consumption):

**1. Energy price**
End customer is free to choose between all sellers that propose delivery in the region you live. Some (typically smaller) sellers are local/regional, some (larger) are national. There is a relatively fierce competition to sign up customers; retailers like Elkjøp will often try to sign you up with an energy seller if you buy something in their store, with tempting "first month free" models etc. 
Each energy seller will normally have several "products" (contract types), there is no real coordination on the pricing models, each seller can "invent" their own. Some have been criticized for making their contracts difficult to compare with competitors. 
The "products" can be:
    - Spot price plus a fixed monthly fee
    - Spot price plus a fixed cost per kWh
    - Combination of the above
    - Fixed price per kWh for coming 3-4 months (period will vary)
    - Fixed price per kWh for coming 12 months 
    - Fixed price per kWh for coming 3 years
    - ...and there might be other variants. 

2. Grid "rental" price
This is the price for transporting the energy from the producer to the consumer, and is paid to a grid company that will always be local/regional. Their fee is strictly regulated by authorities, but will vary between grid companies, depending on their current an future income needs (to build and maintain the grid). There are around 250 grid companies in Norway; smaller and larger. 
The grid company fee has two parts:
    1. Energy part
    This is a fee per kWh, determined by central autorities
    2. Capacity part
    This is a fee that will vary from month to month depending on whether your peak consumption in the preceding month has surpassed certain thresholds. To calculate your peak energy for a month, you take the average of **the three highest-consumption hour-periods in three different calendar days** (24-hour periods from 00:00 to 24:00).
    If that value for the preceding month surpasses threshold 1, you will pay fixed fee A.
    If that value surpasses threshold 2, you will pay fixed fee B.
    And so on. 
    There are typically 4 thresholds you can surpass, each having its own "penalty" (fee). 
    Each grid company can decide their own threshold values, it is not standardized.
    Each grid company can in principle have a different "penalty" (fee) for each threshold level.
    The "Tariff peaks" graph in our firmware tracks the Peak energy consumption according to these rules. 
    This is my current graph:
    [image]
    From this I can read which 3 dates I have the highest whole-hour consumption so far this month. The green dotted line is the average of those 3: 5.7 kW in my case. The red line at 10 is the next threshold that I do not want the green line to reach.

An API has been developed by the grid companies in common - to attempt to simplify the calculation of the **Grid rental price**. However, we see that the inputs (from the grid companies) are littered with errors - so for the moment it that API is basically useless. 

We do not know of any API that can deliver **Energy price**.

VAT (25%) is always added to the bill.

PS
There might be (hopefully) minor errors here - as this is indeed complex. 
My fellow Norwegians: Feel free to correct if you see any errors.

PerBuch54, 2023-09-20:
That confirms my idea that a fixed addon plus the vat calculation could bring the price close to the real price. But let us hear from Sweden.

ArnieO, 2023-09-20:
> That confirms my idea that a fixed addon plus the vat calculation could bring the price close to the real price. But let us hear from Sweden.

Depends on your definition of "close" - but it will definitively be **closer**! 😉 

(And note the edit at the end of my previous post.)

PerBuch54, 2023-09-20:
In DK we had 4 different timezones during the day with different transmission tarifs. But if that are only Denmark i can live with the fixed addon + vat. The 4 timezones will probably be back next year.

KennethLavrsen, 2023-12-05:
Per. The suspension must be something unique to your provider. Most of Denmark still have the tariffs changing 4 times per day. I am in the Radius area.

zelbanna, 2023-12-06:
In my region (SE3) in Sweden we typically have a lot of fixed fees and taxes (emission taxes, regular tax, tax on tax etc.. it is Sweden after all) per kWh and then 25% tax on import rate. In addition to a purely - fixed - montly fees based on fuse size, subscription etc.

For export we don't really get the tax return as private citizens (at the moment we get something back when doing yearly taxes)

Where I live we also have an additional peak load tax based on usage between 9-19 on business days - calculated montly and varying over the year so just forget about that....

PerBuch54, 2023-12-07:
> In DK we had 4 different timezones during the day with different transmission tarifs. But if that are only Denmark i can live with the fixed addon + vat. The 4 timezones will probably be back next year.

I an in N1 area. The 4 timezones will be back next year. I could just add the average to the price and i would bee much better. Just a fixed price add and a factor for vat.

ArnieO, 2023-12-07:
The various inputs here illustrate why we have not tried to implement a cost model that takes into account the grid cost. It is just too complicated, due to the many, many price models used out there.

Norwegian grid companies have worked on a common API, but it is still basically useless due to inconsistent and/or missing data ("rubbish in - rubbish out" syndrom) - AND it would require obtaining API key from each grid company...

KennethLavrsen, 2023-12-07:
Someone will ask - how does the HA sensor look like that uses the script I just posted

here they are

```
  - name: "Electricity Price Now"
    unique_id: electricity_price_now
    unit_of_measurement: "DKK/kWh"
    state: >
      {% from 'tariff.jinja' import radiustariff %}
      {% set tariff = float(radiustariff(0)) %}
      {{ ( float(states('sensor.kamstrup_prices0'))*1.25  + tariff ) }}
      
  - name: "Electricity Price Next Hour"
    unique_id: electricity_price_next_hour
    unit_of_measurement: DKK/kWh
    state: >
      {% from 'tariff.jinja' import radiustariff %}
      {% set tariff = float(radiustariff(1)) %}
      {{ ( float(states('sensor.kamstrup_prices1'), 0)*1.25  + tariff ) | round(2) }}
      
  - name: "Electricity Price In Two Hours"
    unique_id: electricity_price_in_two_hours
    unit_of_measurement: DKK/kWh
    state: >
      {% from 'tariff.jinja' import radiustariff %}
      {% set tariff = float(radiustariff(2)) %}
      {{ ( float(states('sensor.kamstrup_prices2'), 0)*1.25  + tariff ) | round(2) }}

  - name: "Electricity Price In Three Hours"
    unique_id: electricity_price_in_three_hours
    unit_of_measurement: DKK/kWh
    state: >
      {% from 'tariff.jinja' import radiustariff %}
      {% set tariff = float(radiustariff(3)) %}
      {{ ( float(states('sensor.kamstrup_prices3'), 0)*1.25  + tariff ) | round(2) }}

```

zelbanna, 2023-12-08:
So why not do like this implement:
1)  fixed + multipler * region_price 
2) (if you feel the need)  a parallell cost model for the grid- in addition. So not one formula but the aggregate two: one simpler for the hourly price and then and another super-complex-grid-trying-to-do-all :-) )

gskjold, 2023-12-25:
Considering this a duplicate of #252 . See this issue for a build with proposed solution

closed 2023-12-25 by gskjold

GaupeLnx, 2024-01-11:
In Norway there is also to be considered the support from the government where you only pay a given percentage of the KWH price when its is above a given threshold.     calculated for every hour separate. so i would  rather see a possibility for a formula instead of a multiplier to the current rate. This formula could then possible also contain the transport part and its dependency of the last average peak use.

gskjold, 2024-01-12:
The government support is a temporary measure, so I am not putting a lot of effort to support that. Please follow #252 for updates on the new pricing configuration