UtilitechAS/amsreader-firmware #627: Pow-U reboots often, with indication "Reason: Vbat power on reset (1/0)"

ArnieO, 2023-08-31:
**Message from the amsleser.no team (@gskjold and myself):**

We are opening this issue in order to obtain two things: 

1. Get feedback from other users that might be seeing the same issue. So if you are seeing this issue on your Pow-U ESP32 device: Please leave a comment with details here, or email us at post@amsleser.no. Please post screenshot of your Info- and Config page, including the header that identifies firmware version. Please also indicate power meter brand and model. 
2. Update those affected on the status of our debugging process.

**Summary**
We have over the last few weeks received a few notifications on Pow-U devices that have started to reboot often, sometimes several times of day. These reports suddenly started to appear on a product that has not been changed since 2022, and on the same PCBA production batch that we have sold since January 23. 

We do not see the issue on our own device, so debugging this is a challenge! 

We currently have 4 reports related to Aidon meters, 1 report related to a Kaifa/Nuri meter. 

The number of reports are few compared to the number of devices sold, but that does not reduce the problem for those affected. 

**What we have tried so far**
- One customer received a replacement unit, but the problem remained.
- One customer rolled firmware back to v2.2.17 (released long before the problem reports started arriving), but the problem remained.

**Some technical background**
There is in principle (by design) only one way the ESP32 can report "Vbat power on reset (1/0)", and that is if the voltage has for some reason dropped below approx 2.85V.

The device has a voltage supervisor chip that controls the ENABLE line to the ESP32 module. It is implemented with a hysteresis, so that:

- When voltage rises, ESP32 is enabled when voltage has passed above approx. 3.2V
- When voltage drops, ESP32 is disabled when voltage has passed below approx. 2.85V

This ensures that the ESP32 never enters the so called Brownout voltage region, which is voltage below 2.8V. If ESP32 reboots because it detected such a low voltage while operating, it will reboot with a different notification, saying it recovered from brownout. 

The Pow-U generates its operating voltage (nominal 3.3V) from the M-bus signal. That signal is at 24 V (34 V in mid-European meters) in between datagrams, and varies between 24 (or 34) V and 12V lower during data reception. Moreover, it uses a 1F super capacitor to keep the voltage stable while the ESP32 pulls current pulses during data transmission. This has proven to be a successful and stable design - until the issue described herein suddenly started to appear.

**Call for assistance**
As we are unable to recreate the issue, we kindly ask for assistance in getting closer to the cause of the issue. 

Our first step will be to confirm whether the issue is indeed voltage drop as described in previous paragraph. The way we intend to get that sorted out will be to make a Test firmware to replace the current firmware on your device. 

The Test firmware will: 

- Monitor the operational voltage (3,.3V) by reading it with maximum time resolution.
- Display actual voltage on a web page with a "reasonable" update frequency (significantly lower than the voltage sampling frequency). 
- Continuously store to EEPROM the lowest detected voltage, so that it can be displayed after a reboot. 
- Display on the web page the lowest read voltage since test started (i.e. since the Test firmware was installed)

The test firmware will be posted by @gskjold in this thread as soon as it is available. 

Those who are willing to participate in the test must be familiar on how to upload new firmware to the device via the USB cable. It will probably **NOT** be possible to go back to normal firmware via OTA one-click upgrade. If this is unfamiliar to you, please do not install the test firmware.

simenovrebo, 2023-09-01:
Hi,

I'm seeing this a lot with my unit from january 2023. Sometimes it stays up for 2-3 days, but most of the time it seems to like several reboots a day. Would be happy to assist with testing.

[image]

[image]

wknutsen, 2023-09-03:
I have this issue randomly. I just started to save vcc and uptime to timeseries db from 1. September. Since then i have have had two reboots one vbat and one software reset.
This is trend from vbat issue with lowest voltage 2.94v
[image]

Is the reboot reason posted to MQTT ? 

Hardware information:

Country: Norway
Meter: Aidon
Encryption enabled: No
AMS reader: ESP32S2 Pow-U+
M-bus adapter (if applicable):
Relevant firmware information:

Version: 2.2.21
MQTT: Yes
MQTT payload type: JSON
HAN GPIO: Unkown
HAN baud and parity: Unknown
Temperature sensors: No
ENTSO-E API enabled: Yes

ArnieO, 2023-09-04:
@wknutsen Thank you for reporting and accepting to run test firmware when it becomes available. 
Could you be so kind as to post screenshot of your config page? I'm in particular curious to see your setting on Power saving.

Miklagarur, 2023-09-04:
Same issue here. Usually reboots 3-4 times a day, but from time to time it will last longer, like much longer, the longest is 4 days. Meter is KAIFA/Nuri MA304H3E with modified cable as recommended. Power option set to Minimum and are showing 3.29v. -48dBm signal strength. My test shows that it will reboot once wifi is not available regardless of chosen settings. 
That said, I really appreciate this excellent product (Pow-U).

wknutsen, 2023-09-04:
This is my config:
[image]

ArnieO, 2023-09-04:
@wknutsen 
> By looking at voltage it most likely a vbat power on reset issue.

Thank you very much for providing the voltage plots, this is very useful for us.

gskjold, 2023-09-05:
@wknutsen That plot was very interesting, thank you. It suggests that the WiFi modem is kept active more often in that period. Do you have anything on the network requesting data (or ping) the device?

I am also wondering if there could be some types of multicast that could affect the device in this manner, but not sure what that could be yet.

gskjold, 2023-09-05:
As a start, I have created a version of the official firmware that has low voltage logging embedded. I think this will be sufficient. The lowest voltage together with timestamp and a clear button can be found on the status page
[image]

Firmware contained inside this zip. Use "firmware.bin" and upload this via status page.
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12520945/esp32s2.zip)

ArnieO, 2023-09-05:
@Miklagarur 
> My test shows that it will reboot once wifi is not available regardless of chosen settings. 

Du you have this option activated, the cause of your reboots are probably then connection issues in your network:
[image]
In that case, you should not see "Reason: Vbat power on reset (1/0)" on the Info-page.
Kindly check and confirm.

wknutsen, 2023-09-05:
@gskjold 
> @wknutsen That plot was very interesting, thank you. It suggests that the WiFi modem is kept active more often in that period. Do you have anything on the network requesting data (or ping) the device?

I have a web page open 24/7 with AMS reader other than that nothing i know of. I think i have a 2.4 Ghz wifi adapter laying around that can be used to sniff traffic. When i have time I will set up a Raspberry Pi to do some sniffing and give you an update.

> I am also wondering if there could be some types of multicast that could affect the device in this manner, but not sure what that could be yet.

I have a couple of google tv and home that probably generates some multicast traffic.

gskjold, 2023-09-05:
Thanks for the quick feedback. It should survive having the webpage open over time, but I will look into if there are any edge cases where it ends up requesting too often from the device.

ArnieO, 2023-09-07:
**UPDATE**

Based on the various feedback we have received, the problem seems to be voltage stability on the internal 3.3V, generated by a buck converter that pulls power from the M-bus signal line. 

We do not yet understand why and how; we're working on digging further into this.

**Remedy strategy**

1. For those who are affected in a way that causes issues, the emergency solution will be to power the device from the USB connector (from a wall wart or a powerbank, depending on situation).
2. One of the largest issues is with those of you who exploit the running energy estimate. Each reboot destroys that estimate. To remedy this, @gskjold will release a patch that stores accumulated value to flash memory, so that the device wakes from reboot with relatively correct estimate. However, each estimate cannot be stored to flash memory, as there is a limited number of read/write cycles for such a device. The implementation will therefore store for each e.g. 1 kWh accumulated energy (or maybe a bit smaller). This means that recovery will be a potentially "bad estimate" - but not as bad as starting from zero. So... far from perfect, but better. 
3. We continue working on finding root cause and fix it.

Miklagarur, 2023-09-07:
> @Miklagarur
> 
> > My test shows that it will reboot once wifi is not available regardless of chosen settings.
> 
> If you have this option activated, the cause of your reboots are probably then connection issues in your network: [image] In that case, you should not see "Reason: Vbat power on reset (1/0)" on the Info-page. Kindly check and confirm.

Hi Arne,
I can confirm that the device will reboot regardless if this option is checked or not, when I power down my wifi router. 
I have an Asus zenWIFI AX router with Smart Connect enabled. To me it seems like wifi is the culprit, once there are over 20 clients and a lot of wifi activity, AMS unit goes into everlasting reboot or at least are not accessible any longer.
Here are my 2.4ghz settings, anything that should be changed perhaps?
[image]

ArnieO, 2023-09-07:
@Miklagarur Thank you for interesting input!
I know @gskjold is more competent than me on the more advanced Wifi settings, so I will lett him respond.

gskjold, 2023-09-07:
> I can confirm that the device will reboot regardless if this option is checked or not, when I power down my wifi router.

Interesting indeed, currently trying to reproduce that.

> Here are my 2.4ghz settings, anything that should be changed perhaps?

All this depends on what you use that network for:
- If the WiFi clients do not need to be able to see and talk to each other, you can enable AP isolation. This prevents wireless clients from sending data to each other, while still allowing them to send data to the internet or wired clients on the network.
- Roaming is generally a bad idea for 2.4ghz, I would keep that off.
- If there are only low bandwidth type of devices (IoT devices only) on the 2.4GHz, you can set DTIM=10 and Beacon interval=1000. This will help for low power devices, but if you have anything needing bandwidth, changing these will impact those negatively
- WMM can be switched off unless you share the WiFi with multimedia devices like chromecast and such.
- Airtime fairness is by many considered a flawed feature and should be disabled.

gskjold, 2023-09-07:
I can also confirm that it will reboot when loosing wifi, of the simple fact that it burns too much power while searching for the AP. This could mean that this whole thing is related to some connection instability.

ArnieO, 2023-09-07:
> I can also confirm that it will reboot when loosing wifi, of the simple fact that it burns too much power while searching for the AP. This could mean that this whole thing is related to some connection instability.

Veeery interesting observation, indeed!

gskjold, 2023-09-10:
I have a firmware for you guys to test, this should be less power hungry when wifi is lost. Hopefully it is enough

[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12569935/esp32s2.zip)

gskjold, 2023-09-10:
By the way, if any of you have a FTDI adapter that fits on the header on the board, there will be debugging there if you enable serial debugging in settings and set it to INFO level. You will find reason for disconnect there.

wknutsen, 2023-09-12:
> I have a firmware for you guys to test, this should be less power hungry when wifi is lost. Hopefully it is enough

PriceAPI don't work in this version. Kinda annoying to test this version over time.

gskjold, 2023-09-12:
Strange, are you supplying your own API key or not?

simenovrebo, 2023-09-12:
I'm seeing this also on the price API, but I got it working by removing my API key and enabling without API key after reboot.

wknutsen, 2023-09-12:
> I'm seeing this also on the price API, but I got it working by removing my API key and enabling without API key after reboot.

Thank you. That solved it

gskjold, 2023-09-12:
You are absolutely right, using ENTSO-E API key does not work with that version, weird... I have reverted a bump in espressif platform version I did earlier, which resolves the issue.

I have added a feature to preserve realtime data between software reboots.
Also added feature that disconnects WiFi for 30s if power drops too low. I'm not entirely sure how this impacts operation of the device, so only test if you are confident you can re-flash with official release version if this goes south.

[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12589914/esp32s2.zip)

gskjold, 2023-09-16:
I have been running this the last few days in an Ubiquiti environment with channel optimization enabled. Earlier the Pow-U was rebooting every night during the channel optimization, but now it has survived the last few nights. Feel free to test this firmware and report back

senspix, 2023-09-18:
#637 is possibly a duplicate of this issue although it is a POW-P1 device

wknutsen, 2023-09-19:
Update using the first firmware in thread with ENTSO-E api key issue. I have a all time high with 7 days uptime. This looks very promising :)

[image]

wknutsen, 2023-09-19:
> Are you able to make a plot of Vcc only, including y-axes? I am puzzled by the Vcc downstep caused by the firmware change, so would like to see which average value it is now at.

[image]

Firmware 2.2.21
[image]

New test firmware
[image]

ArnieO, 2023-09-20:
@wknutsen Thanks a lot, very interesting. The difference is not large, but it is a distinct difference in both level and variation of the measurements. 

I'll run some tests with the same two firmware versions and see if I can recreate this.

enedberg, 2023-09-21:
I am also having this issue

Pow-U was just rebooted because of a change in the settings ("Auto reboot on connection problem"), but have been showing the "Vbat power on reset (1/0)" error. I have noticed wrong readings in HA so I expect this problem to have started earlier this summer.

[image]
[image]

gskjold, 2023-09-21:
I made some final adjustments that can be tested here: #641

gskjold, 2023-09-29:
Quick summary from v2.2.22 testing:
This issue will be fixed for most cases in this version. Disabling 802.11b will further reduce the chance for reboot. Some still have issues, which we will continue to monitor and investigate.

pomok, 2023-09-30:
Observation before trying #641:
Voltage displayed is around 3.24-3.29V, readings about 3.27V seems to be the norm.

[AMS reader b9d6f16] Up 6 days - The highest uptime for quite some time
Chip: esp32s2 (240MHz)
Device: [Pow-U+]
Last boot: 24.09.2023 03:41
Reason: Software reset (3/0)

Manufacturer: Kaifa
Model: MA105H2E

ArnieO, 2023-09-30:
> Observation before trying #641:

v2.2.22 is released now - so you can just update it directly!

My status:
Pow-K+ on a Kamstrup. I am used to seeing several reboots per day, that I have struggled to understand. I logged Pow-K Vcc on the board during operation (using a voltage logger) and found unexplainable voltage drops lasting up to 10 seconds. 

Now:
Since installing v2.2.22 and disabling 802.11b legacy rates: Still no reboot after 20 hours. 
So this looks very promising!

senspix, 2023-10-01:
More than 48 hours uptime so far! Looking good

ArnieO, 2023-10-01:
My Pow-K too has not rebooted since the upgrade Friday afternoon. 😃

pomok, 2023-10-01:
Have not upgraded to .22 yet. Old test version has gone stable on me...

[AMS reader b9d6f16] -- Up 7 days
Chip: esp32s2 (240MHz)
Device: [Pow-U+](http://192.168.10.50/vendor)
Last boot: 24.09.2023 03:41
Reason: Software reset (3/0)

Manufacturer: Kaifa
Model: MA105H2E

bmork, 2023-10-02:
I see that we already have Vcc plots here, but just to add another data point (running v2.2.22):
[image]

Still missing the uptime plot, but the dips do of course correlate with
```
Last boot: 02.10.2023 04:06
Reason: Vbat power on reset (1/0)
```

The level changes puzzled me too.  They happen on every reboot.  My best guess is that there is some boot time calibration ending up with slightly different results here.

ArnieO, 2023-10-02:
> The level changes puzzled me too.

Thank you for the plot!
There is a phenomena we see that is maybe related to this. We see it when we connect the devices to a current measurement system (Nordic Power Profiler II) that shows us dynamically the power consumption with great resolution.

The phenomena is: Almost systematically, the device slightly changes "consumption profile" with each reboot, giving a distinctly different average consumption. This could indeed be the reason for the (small) voltage difference.

For the moment we do not understand this - and are in an "information collection mode" regarding this phenomena. We see that it is HW-independent (we see the same on all our products). 

If anyone in the community with hands-on experience has already seen something like that before - it would be great if you could share your experience with us!

The cause could in principle be "anything", from HW error in the ESP, to library error from Espressif, to error in Arduino implementation. Or some "obscure" setting that we have missed somewhere. We find it very peculiar that it "toggles" for each reset.

dbeinder, 2023-10-02:
The ADC in ESP32s are notoriously bad, I wouldn't be surprised to see its output change with every reboot. But do you really see the voltage level change when measured independently? What chip do you use as a voltage regulator? The average consumption change can't be more than a 1mA, surely? And ~1mV/mA load dependency seems way too high for any voltage regulator made this century.

ArnieO, 2023-10-03:
Hi @dbeinder, I appreciate your comments!
The puzzling topic of different current draw modes deserves a dedicated discussion thread, so I started one now, please go to https://github.com/UtilitechAS/amsreader-firmware/discussions/648

I prefer to use fixed-voltage LDO in these design, and do not want to "stretch margins" by increasing operating voltage to 3.6V. So thank you for the idea - but ... no.

ArnieO, 2023-10-04:
Still no reboot since I installed v2.2.22 and disabled 802.11b legacy rates (29th Sep).
Pow-K, TP-link Deco M5 mesh network.

I will now enable 802.11b legacy rates to verify whether it is this setting that makes the difference.

wknutsen, 2023-10-04:
A quick update using 2.2.22 with disabled legacy rates.
I just had a vbat incidence. Hope this was just a unlucky coincidence. Voltage seems more stable with this release.
I have been away this weekend and checked status page yesterday last reason was software reset.
[image]

ArnieO, 2023-10-04:
> I will now enable 802.11b legacy rates to verify whether it is this setting that makes the difference.

My Pow-K rebooted with "Vbat power on reset (1/0)" only 1 hour after I enabled 802.11b legacy rates.
So 802.11b legacy rates is definitively what has made my unit reboot frequently. 
"Disabled" it is, then.

simenovrebo, 2023-10-08:
Seems like disabling 802.11b legacy rates did it for me too. I don't think I've seen 6 days uptime ever before on my Pow-U+. 😊

gskjold, 2023-10-14:
It's good to hear that the issue is mostly resolved. For those who had reboot issues resolved by v2.2.22, could you please also checked attached firmware to confirm that this is still the case before I release v2.2.23

[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12903111/esp32s2.zip)
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12903112/esp8266.zip)

Changes:
- Disabled auto disconnect on low voltage
- Added option to disable 802.11b on ESP8266

dbeinder, 2023-10-14:
@gskjold Have you verified this works on ESP8266? When I looked into it, I figured the API you used does not disable 11b, since there are only 3 options: 
[image]
The default has to be `PHY_MODE_11N`. But the default behavior is b/g/n support, so figured `PHY_MODE_11G` is simply going to limit to b/g.

https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf

Looking over it again, it should actually be doable using `wifi_set_user_rate_limit` and `wifi_set_user_limit_rate_mask`. As I read it, they doesn't allow you to get rid of 11b completely, but you can still disable the problematic 1 & 2 Mbps rates in 11b mode which should work just as well.

I really like my [cheap Mikrotik AP](https://mikrotik.com/product/RBmAPL-2nD) to test these things, since you can configure almost any aspect, drop the WiFi signal by 40dB by closing your fist around it and get realtime client stats: 
[image]

gskjold, 2023-10-14:
I have had mixed on this, but I found a few references where setting 11g resolved power and connection issues, so I decided to test it.

> Looking over it again, it should actually be doable using `wifi_set_user_rate_limit` and `wifi_set_user_limit_rate_mask`

I was looking at the same thing earlier, but decided to test setting phy mode, but I think you are right, we should be using this, I will run a new test a bit later today.

bmork, 2023-10-14:
> For those who had reboot issues resolved by v2.2.22, could you please also checked attached firmware to confirm that this is still the case before I release v2.2.23

Not completely resolved for me, but definitely improved by v2.2.22.  A rough guesstimate is that I have about twice the uptime I had before - now typically a day or two.

I just installed the version you attached.  Will let you know if there are any significant changes.

bmork, 2023-10-14:
> I just installed the version you attached

That wasn't entirely true, but **now** it is installed.

And I learned something new about powering the Pow-U+. Simply borrowed the USB-C power plug and cable I use to power my laptop for the upgrade.  The power supply delivers 3A at all the lower voltages, which obviously should be plenty.  But it wasn't.  The Pow-U+ rebooted unexpectedly several times while trying to upload the new firmware.  I guess the voltage drop in 3m USB-C cable is too much for the Pow-U+ regulator.

Or was the problem using a PD power supply? Maybe that's a bad idea?

Used a 15cm USB-A to -C cable from my laptop instead, like I've done before, and everything was fine.

ArnieO, 2023-10-14:
> And I learned something new about powering the Pow-U+. Simply borrowed the USB-C power plug and cable I use to power my laptop for the upgrade. The power supply delivers 3A at all the lower [...]

Thank you for mentioning this, but we never heard of any such issue before. 

And even if there is a quite strong inrush current spike (supercap charging), it is well below 3A.
The USB-C socket on Pow-U is implemented in full accordance with the standard: https://www.usb.org/document-library/usb-type-cr-cable-and-connector-specification-release-22
It has a "simple" termination to indicate to a PD power that it wants 5V, no advanced voltage or current level negotiation. 
To be exact: There are two 5k1 resistors, connected to the CC1 and CC2 terminals of the USB-C socket.

I find it very unlikely that cable voltage drop is the issue. Maybe there is some very "nervous" protection circuitry in your computer - that reacted to the initial inrush current?

Anyway: Interesting to know this - in case anyone else runs into the same issue. Thank you!

bmork, 2023-10-14:
> I find it very unlikely that cable voltage drop is the issue. Maybe there is some very "nervous" protection circuitry in your computer - that reacted to the initial inrush current?

The problem happened when powering from a 65W USB PD supply, not from the laptop.  I doubt the supply will limit the current until it reaches 3A.  I didn't disconnect the Pow-U+ from the HAN port at all so the supercap was already charged.  And the reboot didn't happen until I tried to upload the new firmware, not when I connected the external supply.

Could have been a co-incidence.  But I did try several times before switching cable and supply, and then it worked the first time. Guess it could be an issue with the supply.  Haven't used it much at 5V, if at all.  It's a bit overkill for that.  Just happened to be at hand.

gskjold, 2023-10-15:
[esp8266.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12909729/esp8266.zip)
I have been experimenting a little bit and changed the way 11b is disabled for 8266, thanks for the tip @dbeinder . At least on my setup, it does not drop into b modes now.

bmork, 2023-10-17:
> The problem happened when powering from a 65W USB PD supply

I've now tested this with two different PD supplies. Both are capable of charging my headset, which I believe supports 5V only too.  Used three USB-C to USB-C cables in the process - all known to be working fine.  The conclusion is that my Pow-U+ isn't powered by PD supplies. Disconnecting it from the HAN port with the PD supply connected makes the Pow-U+ fall off the network. And the LEDs dim slowly as the supercap is discharged.  So that explains the upgrade issue.

External power works fine with a 5V charger or USB port as source, using a USB-A to USB-C cable. But it does not work with PD.

Not sure if this is completely off topic here, or if this is a hardware bug possibly related to the topic?

ArnieO, 2023-10-17:
> I've now tested this with two different PD supplies. Both are capable of charging my headset, which I believe supports 5V only too. Used three USB-C to USB-C cables in the process - all known to be [...]

Interesting indeed, as this is a potential (but surprising!) HW bug. 
(@gskjold : Can you help me move this to a separate topic? It seems totally unrelated to the subject here.)

I will re-read the USB Type C documentation and see if I can find a reason for this - and a solution for next layout. Thank you for your assistance, @bmork !

ArnieO, 2023-10-17:
> I will re-read the USB Type C documentation and see if I can find a reason for this - and a solution for next layout.

I confess I have not re-read the [USB type C standard](https://www.usb.org/sites/default/files/USB%20Type-C%20Spec%20R2.0%20-%20August%202019.pdf) (>370 pages), I have re-read a number of web sites explaining how to do the common stuff.

And what I find everywhere is the same. To indicate to the USB type C power supply that the device is an upstream-facing port (UFP) that needs 5V up to 3A, the solution is what we have implemented: a 5k1 resistor to GND on each of the CC terminals. 

I found this page has quite precise information on how this works, and how to implement a device with UFP: https://dubiouscreations.com/2021/04/06/designing-with-usb-c-lessons-learned/

HOWEVER, I now also found this forum thread: https://forum.digikey.com/t/simple-way-to-use-usb-type-c-to-get-5v-at-up-to-3a-15w/7016

See especially the post written by "mvduin", that goes in more detail. And what I understand from this is that a PD source (DFP) will signal how much the device needing power (UFP) how much current it is allowed to draw, by monitoring the voltage on the two CC pulldown resistors. And IF the DFP (source) indicates a lower current level than the device needs, it is the responsibility of the device to either change its consumption OR shut down (disconnect from the source). 

@bmork: My understanding is then that your PD power for some reason does not accept delivering the current needed by the Pow-U, and signals that by the current it runs through the CC resistor - which the presently used design of the Pow-U does not act upon. The result is that Pow-U continues to pull as much current as it needs - and as a result probably pulls down the bus voltage to a level where the Pow-U is unable to operate.

What puzzles me is that you say this happened on a 65W PD supply.

@bmork: I apparently have to buy a PD supply to test this myself, but in the mean time: Are you equipped with a multimeter so that you can probe the voltage of the CC resistors, if I indicate to you where to probe? It would be very interesting to see what the voltage is, which could help understanding what is going on here. 

**Conclusion**
Our USB type C interface is not fully compliant to the USB standard, as it does not correctly handle PD supplies that signal a lower current supply capability than the device needs. To become fully compliant, an USB Type-C controller chip has to be included, as well probably a switch component (I'll have to study this further). Given the available PCB area on Pow-U, this seems like a significant modification that will **_not_** be introduced on the next (soon upcoming) relayout. So until further, we should signal in the User Manual that our products can **_not_** be powered from Type-C PD power supplies.

bmork, 2023-10-17:
> What puzzles me is that you say this happened on a 65W PD supply.

Yup.  The other PD supply I tried is 48W.  Both specify 3A capability @ 5V.

> Are you equipped with a multimeter so that you can probe the voltage of the CC resistors, if I indicate to you where to probe?

Can do. I measured between the USB-C connector shield and the R7/R8 end closest to the connector.  Is that correct?  This gave 1.62V on R7 and 0 on R8. And vice versa after turning the USB-C 180 degrees.

For the fun of it I tried a few other USB-C power sources.  My Android phone powers the Pow-U+ without complaints.  A HP laptop with USB-C ports does not.  The 100W USB PD battery bank I have was most fun  It turns on when I connect the Pow-U+, as it should.  But it looks like it is using the Pow-U+ as a power source - the supercap driven LEDs dim much faster after connecting to the battery bank.

(the battery bank should also deliver up to 3A @ 5V)

> we should signal in the User Manual that our products can not be powered from Type-C PD power supplies.

FWIW, I don't see this as a problem.  The USB-C connector is convenient in any case.  And I would never use anything but a dumb 5V supply for permanent powering in any case.  Only noticed this issue becasue the PD supply was right next to me and I only needed it for the 1 minute firmware upload.  Wouldn't have thought about using a PD supply otherwise.

If the RaspberryPi 5 can use a non-standard USB-C supply, then....

ArnieO, 2023-10-17:
> I measured between the USB-C connector shield and the R7/R8 end closest to the connector. Is that correct? This gave 1.62V on R7 and 0 on R8. And vice versa after turning the USB-C 180 degrees.

Yes, correct. So this is even more puzzling. 
According to this table (from the 2nd link in my previous post), the PD power then says you're allowed to draw up to 3A. 
[image]

What is the voltage on testpoint TP1 when PD power is connected and the unit does not work? This is the voltage delivered from USB.
[image]

bmork, 2023-10-17:
> What is the voltage on testpoint TP1 when PD power is connected and the unit does not work? 

2ish and falling.  Which is the same I measure when disconnecting the power supply.

bmork, 2023-10-17:
Hey, think I'm onto something.  I measured on the USB-C connector pins and see that you leak this voltage out on VBUS.  This is probably the issue, hitting some protection mechanism.  I tried discharging the supercap completely before connecting the PD supply, and then it worked!  With 5V on TP1 obviously.

[16 more replies]

closed 2023-11-12 by ArnieO