UtilitechAS/amsreader-firmware #628: Improve scaling of graphs with very unequal import/export numbers?

ThorAlex87, 2023-09-01:
**Is your feature request related to a problem? Please describe.**
When graphs have both import and export they scale with zero in the middle and an equal scale over and under zero. With my new solar setup and home assistant moving loads to when there is high solar production I get a very large difference in the bars over and under zero, causing a lot of wasted space in the import graph in my case.

**Describe the solution you'd like**
Maybe allowing the zero line to move up and down would be better? Might have to make the zero line a little thicker too.

**Describe alternatives you've considered**
Dunno, that's my best idea. I can see the advantage of leaving it as is too, in that it shows off the difference in import/export more obviously. (Leaving zero in the middle and scaling differently for import and export is another option, but a bad one IMO.)

**Additional context**
Screenshot this morning, highest hourly import last 24 hours is 0,5kWh but graph scales to 6,2kWh due to the export numbers. In my case the export could potentially get close to 15kWh on a good day with import peaking as low as 0,2kWh in the same graph, resulting in the space above zero being almost completely empty.
[image]

ArnieO, 2023-09-01:
Thank you for your comment and proposed solution!
I think your idea of de-centering the zero line would be an improvement. 
Now let us hear what @gskjold thinks, and whether it is implementable in the graphics tool we're using.

gskjold, 2023-09-05:
For next patch:
[image]

ThorAlex87, 2023-09-06:
That looks great, thanks! Is there an estimated ETA on the next patch?

gskjold, 2023-09-06:
I would guess within a couple of weeks, as there are enough changes ready to go.

spacecosmos, 2023-09-18:
Maybe an additional possible solution could be to have different colours on the graph for import and export? Green on import perhaps? Just my two cents.

ArnieO, 2023-09-18:
> Maybe an additional possible solution could be to have different colours on the graph for import and export? Green on import perhaps? Just my two cents.

I definitively think that is an idea worth pursuing! 👍

gskjold, 2023-09-21:
Final testing of this feature in issue #641 

Also see #544 regarding colors on graphs

closed 2023-09-21 by gskjold