UtilitechAS/amsreader-firmware #630: One hour delay between AMS reader and Home Assistant

PalmMx, 2023-09-06:
**Issue description**
My Pow-K is working fine, but as I use its data mostly in Home Assistant, I find it very confusing and annoying that there seems to be an one hour delay between the AMS reader and the energy dashboard in Home Assistant. Please see attached pictures.

Tonight at 19:50 I saw the value of 1.5 kWh for the time slot 17:00-18:00 in AMS reader, while the corresponding data in Home Assistant was registered as the 18:00-19:00 time slot. Also the last hour shown in the AMS reader picture, i.e. 18:00-19:00 amounting to 1.0 kWh, was not registered at all at that moment by Home Assistant.

I have the same time zone settings in both AMS reader and in Home Assistant: Europe/Stockholm (GMT+1)

As I do calculations in Home Assistant based on both energy consumption and energy price per hour I get messed up values as the energy consumption for the last hour does not match the energy price for that specific hour.

Home Assistant uses the total energy meter value and that is only updated once per hour, can that update frequency be increased from the Pow-K?

Or is there any other known workaround available for this issue?

**Hardware information**
 - Country: Sweden
 - Meter: Kamstrup
 - AMS reader: Pow-K (GPIO12)
 - Chip: esp8266

**Relevant firmware information:**
 - Version: 2.2.21
 - MQTT: yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO: GPIO12
 - HAN baud and parity: 2400 8N1
 - ENTSO-E API enabled: yes

[image]

[image]

gskjold, 2023-09-07:
This is an unfortunate effect of your meter reporting the accumulated values only once per hour, which happens 15s over the hour for you. AMS reader will only report actual values, except for the realtime sensors which are estimated.

Duplicate of #300 

Maybe someone in the community have solved this in HA somehow, but this is not something we will try to work around in the firmware.

ArnieO, 2023-09-07:
I have an idea of a _potential_ solution, although I am not sure if anyone has tried it yet.

The grid company will normally have a software (Omnisoft VisionAir) where they can communicate with the meter via the "backbone" wireless network that is used for reading the meter. In this software they can also select between various payload formats. There is a screenshot from that software on page 17 of [this document](https://documentation.kamstrup.com/docs/HAN_NVE/en-GB/Information_sheet/CONTAAA62C8B611D4C4EB3186FD81AEE0152/), which is the Kamstrup Omnipower HAN port Interface Specification, where the selectable payload formats are indicated:
[image]

Power meters in Norway and Sweden are configured with the "HAN-NVE" payload, while the same meters in Denmark are configured with "DLMS push 1". The advantage with "DLMS push 1" is that all payloads are identical, and include accumulated readings (kWh and kVAr). Chapter 5 of the same document explains which data points are transferred in the three optional payload formats. 

So a potential solution could be to contact the grid company and request them to reconfigure the power meter to send "DLMS push 1" in stead of "HAN-NVE".

I assume that some grid companies will be more service minded than others - but it could be worth the phone call to try!

 It would be nice if those that attempt this could report back here in this thread.

StarkJohan, 2023-09-07:
Sounds very promising and I will surely contact my grid owner. Before I do that though; am I correct in my assumption that this data format also is fully supported by the software today?

ArnieO, 2023-09-07:
> I see an entry for DLMS in the Wiki so I feel pretty confident. Please confirm anyway.

Sorry, I forgot to write that. 
Yes, I confirm: The firmware reads whatever is the content of the payload.
You'd be surprised to see the number of variants between power meter manufacturers and models...!

StarkJohan, 2023-09-07:
I've asked them, we will see how they handle it.

ArnieO, 2023-09-07:
There is a small snag regarding the P1 protocol: The meter does not push data unless a pin on the 6-pin connector is pulled low. On the current generation of Pow-K, that pin is not exploited on our board (not connected). On the next generation it will be fed from a GPIO on the microcontroller. 

In order to use the current Pow-K with a P1-configured meter, you will need to do a small modification of the board, as described here: https://github.com/UtilitechAS/amsreader-firmware/issues/503#issuecomment-1710195716

ThorAlex87, 2023-09-07:
You can try using the "Current hour used" sensor instead, that one is handled correctly by home assistant. I think it's calculated by the firmware? Or you can make a Riemann sum sensor in HA to calculate the hourly consumtion, this also works but is a little more complicated. 

There is a bug with the "current hour" sensors when exporting power that will be fixed in the next patch, but it worked fine for me until I got my solar setup.

StarkJohan, 2023-09-07:
I've tested the HA Riemann sum earlier and while it works pretty well, I do prefer actual data if I can get it. 
The "hourused" sensor must have been introduced rather recently and is likely based on the same, or very similar, calculations. I know we've discussed this exact topic before and at that point in time this sensor was not present. It's a very helpful value when true data is absent.

I haven't been following the development in a while as my ams reader is extremely stable and just works. In all aspects it's a "set it and forget it" device to me. Excellent work!

bardahlm, 2023-09-08:
See also #394

StarkJohan, 2023-09-11:
Just recieved an email from my grid company that my meter now is reconfigured to send DLMS data. I checked and now I have mqtt updates including totar accumulated import every 10 seconds!

Fantastic customer service from my grid company! Excellent tip from @ArnieO

ArnieO, 2023-09-11:
> Just received an email from my grid company that my meter now is reconfigured to send DLMS data.

Good service from them, yes! 
I have seen only one case so far where the grid company refused such a request - so I think most of the grid companies are kind enough to do it.

ThorAlex87, 2023-09-11:
> 

Nice to know there is a solution, I'll have to try this! 
Could you post the text of the e-mail you sent them? My grid company (Tensio TN) has never been the most helpful, so having a "known to work" template for an e-mail would be good!

ArnieO, 2023-09-11:
I would have tried the good old telephone, but that's me... 😅

StarkJohan, 2023-09-11:
> > 
> 
> Nice to know there is a solution, I'll have to try this! Could you post the text of the e-mail you sent them? My grid company (Tensio TN) has never been the most helpful, so having a "known to work" template for an e-mail would be good!

I spoke to their customer service on the phone before sending them more information. I wrote nothing special but I referred to the manual and sent a screenshot of the post from @ArnieO above. Never hurts to be really nice :-)

"Hej igen!

Pratade nyss med XXXXX gällande vilket datafortmat jag får ut på min HAN-port och att jag hade önskat byta till DLMS-formatet. Det gör mig veterligen ingen skillnad för er funktion men min blir betydligt bättre med fullständig data var tionde sekund istället för var sextionde minut.

Här är lite mer kött på benen. Se bilden för info om valen och länken för dokumentet som refereras till (manualen till mätaren).

https://documentation.kamstrup.com/docs/HAN_NVE/en-GB/Information_sheet/CONTAAA62C8B611D4C4EB3186FD81AEE0152/?page=1

Tack för att ni är öppna för förslag och för trevligt bemötande som alltid.
Stort tack på förhand."

ThorAlex87, 2023-09-11:
Thanks! 

If I call them they are very helpfull and understanding,  but ask to write an e-mail and say they'll get back to me in a few days. Then it usually takes weeks and several follow-up e-mails to get any sensible answer out of them... I'll look into it once my current grid over-voltage issue is sorted. After all, it's only been 12 days of complete silence since the call and e-mail.

closed 2023-10-04 by gskjold

SVH-Powel, 2024-02-09:
> which is the Kamstrup Omnipower HAN port Interface Specification, where the selectable payload formats are indicated

@ArnieO Do you know if the same apply to Aidon meters?

ArnieO, 2024-02-09:
> @ArnieO Do you know if the same apply to Aidon meters?

I have not seen any published information on how their "backstage" system is set up, so our source of information is the Interface specifications.

During our development of Pow-P1 (for meters configured to the "P1 companion" standard), we had helpers (testers) in Sweden that were able to switch between ASCII and binary format on their meter by contacting their grid company. But for P1 meters, the Aidon interface specification mentions several alternative payload formats. 

The Aidon interface spec for meters with HAN-NVE interface only mentions the NVE format, so I assume (but do not know for sure!) that their firmware might not allow switching between alternative payload formats. 

See for instance this version of the spec (it might not be the newest available, however):
https://github.com/UtilitechAS/amsreader-firmware/blob/main/doc/Norway/Aidon-HAN-Interface-Description-v11A-ID-34331.pdf

SVH-Powel, 2024-02-09:
> See for instance this version of the spec (it might not be the newest available, however):

I found version 1.7 of this document from 2022 and they does not mention anything else than han-nve.

ArnieO, 2024-02-09:
> I found version 1.7 of this document from 2022 and they does not mention anything else than han-nve.

Yes, I did not mention I too have seen newer versions - and they are almost identical.

Norleif, 2025-12-02:
A bit late to the party but could be helpful for others searching for this. Maybe add it to the wiki.
In the Home Assistant database, adding a trigger to catch the insertion of the "Accumulated active import" and modifying the timestamp sligthly backwards fixes the one hour delay.

- Start by finding the entity ID for "Accumulated active import". In my case: "sensor.ams_2088_tpi".
- In home assistant database table "states_meta" find the "metadata_id" for your instance of the "tpi" sensor. In my case 81.
- Determine a reasonable time amount to subtract from the time stamp. In my case 15 seconds has been enough.
- Add the following trigger to your database substituting my values by the ones matching your system:
`CREATE TRIGGER energy_skew AFTER INSERT ON states FOR EACH ROW WHEN (NEW.metadata_id = 81) BEGIN UPDATE states SET last_updated_ts = NEW.last_updated_ts - 15 WHERE state_id = NEW.state_id; END`

To get access to the Home Assistant database, I used this addon:
https://github.com/hassio-addons/addon-sqlite-web/