UtilitechAS/amsreader-firmware #631: Every so often I need to pull the Pow-K to get it back reading data

MTrab, 2023-09-10:
**Describe your problem**

Every so often (could be weeks or months in between) I need to physically remove the Pow-K from the Kamstrup meter to get it back reading data, otherwise I only get ` No HAN data received last 30s, single blink`

**Hardware information:**
 - Country: Denmark
 - Meter: Kamstrup Omnipower
 - Encryption enabled: No
 - AMS reader: Pow-K
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.21
 - MQTT: Yes
 - MQTT payload type: RAW
 - HAN GPIO: ?
 - HAN baud and parity: 2400 8N1
 - Temperature sensors: None
 - ENTSO-E API enabled: No

ArnieO, 2023-09-11:
Thank you for reporting!
Have you set this option in the config?
[image]

MTrab, 2023-09-11:
Yes - that have been set from the beginning.

Today it seems like it have entered a boot-loop. I can get contact for a few seconds and then it reboots.

ArnieO, 2023-09-11:
> Today it seems like it have entered a boot-loop. I can get contact for a few seconds and then it reboots.

That is not good, it sounds like your card has an issue.
We take pride in having all our sold units up and running, so I would like to send you a new card. You will then get the newest version with an ESP32-S2 module. 

Please contact me on post@amsleser.no and confirm your shipping address.

If you are quick, I may be able to get it mailed today (our freight company Bring has some website issues now, so the batch due for delivery at the post office today is a bit delayed).

MTrab, 2023-09-11:
Mail sent - and thank you

closed 2023-10-04 by gskjold