UtilitechAS/amsreader-firmware #632: Wrong current month peaks indexing

longer83, 2023-09-10:
**Describe the bug:**

Wrong current month peaks indexing. Current month peaks show incorrect values, peak 3 always "unknown".

**Screenshots**

[image]
[image]

**Hardware information:**

 - Country: Norway
 - Meter: Kamstrup Omnipower (DK-8660)
 - Encryption enabled: no
 - AMS reader: Pow-K+

**Relevant firmware information:**

 - Version: 2.2.21
 - MQTT: yes
 - MQTT payload type: HomeAssistant
 - HAN baud and parity: 2400 8N1
 - ENTSO-E API enabled: yes

**Additional context:**

Debug log:
```text
name: Current month peak 3
state_topic: amsleser/realtime
unique_id: ams-xxxx_peaks3
object_id: ams-xxxx_peaks3
unit_of_measurement: kWh
value_template: '{{ value_json.peaks[3] | is_defined }}'
device:
  identifiers:
    - ams-xxxx
  name: AMS reader
  model: Pow-K+
  sw_version: v2.2.21
  manufacturer: amsleser.no
  configuration_url: http://ams-xxxx.local/
device_class: energy
platform: mqtt
```
```text
Received 15:43:32
QoS: 0
Payload: max: 3
peaks:
  - 3.57
  - 2.45
  - 3.06
threshold: 5
hour:
  use: 1.38
  cost: 0
  produced: 0
  income: 0
day:
  use: 21.9
  cost: 0
  produced: 0
  income: 0
month:
  use: 63.5
  cost: 0
  produced: 0
  income: 
```

Looks like the indexing is incorrect and peaks should have following indexes:

```text
value_json.peaks[0]
value_json.peaks[1]
value_json.peaks[2]
```
not
```text
value_json.peaks[1]
value_json.peaks[2]
value_json.peaks[3]
```

Issue has been also raised up on amshan-homeassistant GitHub: [#65](https://github.com/toreamun/amshan-homeassistant/issues/65).

gskjold, 2023-09-11:
Nice catch, thanks!

gskjold, 2023-09-21:
Final testing of this feature in issue #641

closed 2023-09-21 by gskjold