UtilitechAS/amsreader-firmware #633: Local API

tux-, 2023-09-11:
**Is your feature request related to a problem? Please describe.**
If for what ever reason one looses mqtt connection for a while, some data may not be posted to user database. Also if one does not wish to use mqtt for some reason (remote hourly logging, etc) it would be nice to have an api to retrieve historical data via API requests.

**Describe the solution you'd like**
Could be fairly simple, just a single rest endpoint that let you query to get hourly import and export for last 48 hours in a json format would be great!

**Describe alternatives you've considered**
Have considered using a raspberry pi with a amshan usb dongle instead.

ArnieO, 2023-09-12:
Please see [Firmware User Manual](https://amsleser.no/index.php?controller=attachment&id_attachment=46) chapter 11. 

Can you use those?

tux-, 2023-09-12:
Interesting! Possibly the dayplot.json endpoint? I du not understand the data format there tho, guessing i## is import and e## = exports. But the numbers does not seem to correlate with what I see in the ui, maybe there is a trick to reading them? Checked the page https://github.com/UtilitechAS/amsreader-firmware/wiki/Message-formats#httpams_hostnamedatajson that was linked to in the firmware manual, but the dayplot.json is missing from the documentation there.

gskjold, 2023-09-12:
I can add some information about the plot json in the wiki, thanks for the heads up.
Regarding the dayplot, you are correct that i==import and e==export. The number is UTC hour for the last 24 hours

gskjold, 2023-09-12:
> so the day-split in in the middle of the line of numbers?

I understand the reaction, its just the way it had to be with how the data storage was constructed from the beginning, while taking time zone out of the equation

gskjold, 2023-11-18:
**Origin Time Worklog**
Estimated: 48.0 hours
Logged: 0.3 hours
Remaining: 47.8 hours

Gunnar Skjold: 0.3

gskjold, 2023-11-18:
Updated wiki: https://github.com/UtilitechAS/amsreader-firmware/wiki/Message-formats

closed 2023-11-18 by gskjold