UtilitechAS/amsreader-firmware #634: HASS: Expose an update sensor

LordMike, 2023-09-11:
**Is your feature request related to a problem? Please describe.**
I was following up on updating devices and realized that the amsleser may not have exposed an update sensor yet. 

It’s be great if it were, so Home Assistant can notify me in turn. 

I’m not sure if this firmware knows updates are available unless a user opens the page, but if it does, you could expose it. 

**Describe the solution you'd like**
An implementation that exposes an update sensor. 

https://www.home-assistant.io/integrations/update.mqtt/

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

pvarlien, 2023-10-23:
Isn't this a duplicate of issue #263?

gskjold, 2023-11-18:
Correct

closed 2023-11-18 by gskjold