UtilitechAS/amsreader-firmware #636: Samsung Family Hub refrigerator built-in browser bug?

spacecosmos, 2023-09-18:
**Describe the bug**
When viewing the AMS reader on the built-in browser on the Samsung Family Hub refrigerator, the Voltage and Ampere usage aren't displayed at all.

**Picture of bug:**
[image]

**Expected behavior**
To display the correct values of Voltage and Ampere usage.

**Additional context**
230V IT grid - 63A main fuse - 21,3 kWp solar panels

gskjold, 2023-09-18:
I love that you have a fridge with a browser! I'm assuming volt/amp graph is displayed on other devices?

spacecosmos, 2023-09-18:
> I love that you have a fridge with a browser! I'm assuming volt/amp graph is displayed on other devices?

Thank - I kinda love it too! :)

On topic: Yes - sorry for not pointing that out, but it works and displays as intended on everything else. So might be a isolated on the Samsung fridge itself I would imagine. But not sure how to properly "debug" the bug. Maybe going to some websites that produces info based on browser type? Please advice if there is anything you want me to test or troubleshoot.

gskjold, 2023-09-18:
I will think about what this could be and how we can best debug it and get back to you

ArnieO, 2023-09-18:
@spacecosmos 
Just to sort out one potential issue: 
Is your fridge software running latest software update? https://www.samsung.com/us/support/answer/ANS00086663/

spacecosmos, 2023-09-18:
> @spacecosmos Just to sort out one potential issue: Is your fridge software running latest software update? https://www.samsung.com/us/support/answer/ANS00086663/

Yes it is. I usually make sure I always run the newest software on all my devices (if no report of serious bugs are reported).

gskjold, 2023-11-18:
Trying some adjustments that might work. I am not sure what kind of device you have, so I'm attaching firmware for all ESP32 types. If you have a Pow-*, unzip esp32s2.zip and upload firmware.bin via the status page
[esp32.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13399582/esp32.zip)
[esp32solo.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13399577/esp32solo.zip)
[esp32c3.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13399579/esp32c3.zip)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13399580/esp32s2.zip)

spacecosmos, 2023-11-18:
@gskjold

**Device information**
Chip: esp32s2 (240MHz)
Device: Pow-K+

This kinda worked! :)

The Voltage seems to be missing, but everything else is now present. So whatever you tried, semi-works. You probably need to tweak it a little bit to make it perfect!

gskjold, 2023-11-18:
Interesting, adding a title to the graph seems to ensure that the graph shows. I'll think about how to best fix this permanently

gskjold, 2023-11-20:
Im trying a new approach, please test: 
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13415908/esp32s2.zip)

spacecosmos, 2023-11-20:
> Im trying a new approach, please test: [esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13415908/esp32s2.zip)

Thanks for this. I tried it, and unfortunately it went back to the old view. Meaning it doesn't show the two Voltage and Ampere values again.

gskjold, 2023-11-25:
OK, hope you are up for some more experimentation, here is another one :)
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13463324/esp32s2.zip)

spacecosmos, 2023-11-25:
> OK, hope you are up for some more experimentation, here is another one :) [esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/13463324/esp32s2.zip)

Thanks! I won't be able to test this until Tuesday actually, as I'm out traveling. But I'll let you know how it went when I get the chance to test it. :)

gskjold, 2023-12-04:
Previous change for amp graph will be added to both in next patch

closed 2023-12-04 by gskjold