UtilitechAS/amsreader-firmware #637: Multiple (= very often) reboots (=short uptime) per day on POW-P1 device

senspix, 2023-09-18:
The max uptime is very often under 60 minutes

In addition, but possibly separate problem: The month plot typically only shows last 10 (not exact) days of data before resetting. 

**To Reproduce**
Just check dashboard regularly

**Expected behavior**
Longer uptime

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware information:**
 - Country: Sweden
 - Meter: Sagemcom T211
 - Encryption enabled: No
 - AMS reader: Pow-P1

**Relevant firmware information:**
 - Version: 2.2.21
 - MQTT: no
 

**Additional context**
See below

amsconfig
version v2.2.21
boardType 6
mdns 1
meterBaud 115200
meterParity 8N1
meterInvert 0
meterDistributionSystem 2
meterMainFuse 20
meterProductionCapacity 11
gpioHanPin 16
gpioHanPinPullup 1
gpioApPin 0
gpioLedPinRed 13
gpioLedPinGreen 14
gpioLedRgbInverted 1
gpioVccPin 10
gpioVccOffset 0.00
gpioVccMultiplier 1.000
gpioVccBootLimit 0.0
gpioVccResistorGnd 22
gpioVccResistorVcc 33
ntpEnable 1
ntpDhcp 0
ntpTimezone Europe/Copenhagen
ntpServer pool.ntp.org
entsoeArea 10Y1001A1001A47J
entsoeCurrency SEK
entsoeMultiplier 1.000
entsoeFixedPrice 0.000
thresholds 5 10 15 20 25 50 75 100 150 65535 3
dayplot 5 1695042000 5941965 0 274 260 246 260 311 269 3289 9669 7804 5783 4266 925 0 0 0 0 14 223 221 242 378 262 263 274 8132432 0 0 0 0 0 0 0 0 0 0 0 2459 2781 7142 5681 3333 584 0 0 0 0 0 0 0
monthplot 6 1694988000 5908072 0 0 0 0 0 0 0 0 0 0 0 0 0 0 2099 41282 3808 2966 0 0 0 0 0 0 0 0 0 0 0 0 0 0 8127192 0 0 0 0 0 0 0 0 0 0 0 0 0 0 27755 47241 47665 0 0 0 0 0 0 0 0 0 0 0 0 0 0
energyaccounting 6 9 0.00 128.42 145.36 0.00 369.22 498.96 12 2.27 13 1.68 18 9.66 8 0.89 15 7.36 116.78 237.10

senspix, 2023-09-19:
Hi,

Yes, I installed it in february. The ESP32 and the board gets power from the Sagemcom device. No other 5V source available. I could put a battery there temporarily but....

**Symptoms**:
1. Even though I've  tried different settings for "Power saving" and "Auto reboot..." I typically see uptimes less than 1 hour. 
2. The month plot regularly loses data, i.e. only the most recent 4-8 days show any data. There is no issue with the price graph or the 24-hour graph.

The Wifi-router (TP-Link) is close by, although inside, and I've changed its setting so that the Amsleser should only be _offered_ 2.4 GHZ. A Wifi RSSI of -39 dBm is definitely OK.

I read all the comments for #584 and that made me check 5V and 3.3 V pins and I measured 5.2 and 3.31 V with a standard multimeter.

I connected with telnet and looked at verbose output but the output seems ok.

The issue with the monthplot is constant now. It did not happen the first months. I don't know if it started with the last update or earlier.

I work with VS Code, PlatformIO with ESP32s (preferably using esp-idf) and other µ-controllers daily so I can try/assist if needed. However, front-end coding is not for me.

Cheers

ArnieO, 2023-09-20:
Thank you very much for the additional information; very helpful!

Regarding the missing data: 
Various brands P1 meters send data at different time intervals; does yours send each second?
Have you tried to increase the buffer size? If not, please increase buffer size to see if that helps. 

Regarding reboots:
Please confirm the "Reason" reported on the (i)nfo page.
And if you can attach a power bank for some time as a test - that could also potentially give some information. 
We have an issue currently being investigated where devices reboot if they loose Wifi connection (Wifi SSID not reachable), because they try too heavily to reconnect (and thus draw too much current. What you see could potentially be related - but since you are not using MQTT, you are probably not able to provide a time plot of Vcc. Anyway, we're working on a fix for that (a less current consuming strategy when SSID connection is lost). 

But please try increasing buffer size first, to see if that impacts the apparent missing readings. 
[image]

senspix, 2023-09-25:
So I
1. Powered the unit with a Powerbank => Improved uptime dramatically
2. Installed version in #641 => Could not see any change

other things
1. I got a `Software reset (3/0)` in the middle of the night while running on battery
2. The energy use diagram is *up to* 11 days now. That has not happened for approx. 2 months.

I will remove the battery later today to see if the reboot often issue reappears.

ArnieO, 2023-09-26:
> 2\. The energy use diagram is _up to_ 11 days now. That has not happened for approx. 2 months.

Sorry for being slow here - but what does this mean?

senspix, 2023-09-26:
Hi, see attached PNG
[image]

ArnieO, 2023-09-26:
[image]

Wow - now I understand; very surprising!

Can you share your Config screen too, please?

gskjold, 2023-09-26:
That issue is fixed in the test releases in #641 

> 2\. The energy use diagram is _up to_ 11 days now. That has not happened for approx. 2 months

I'm assuming this is what you are referring to? The test version fixed that problem for you?

senspix, 2023-09-26:
Two screens: info and config. BTW, now seeing record uptime of 9 hours !? Not on battery power!

Since there is more data in the graph now it might be the case that the the release in #641 fixed it.

Also. it's not just the graph that is missing the data. A JSON export also lacks data.
[image]
[image]

senspix, 2023-10-01:
It seems like this, at least the reboot issue, was fixed in 2.22.22. #641.  Great work, thanks. 
I'll come back with new issue if the 30-day energy graph continues to misbehave.

closed 2023-10-01 by senspix