UtilitechAS/amsreader-firmware #638: Wrong number shown for today in "Energy used last 31 days"

cmock, 2023-09-19:
**Describe the bug**

I'm seeing a bar for today's imported energy in the "Energy used last 31 days" graph, which I think shouldn't be there. 
The bar is also wrong, it does not reflect the real imported energy (bar shows 4.2 kWh, table at the top shows 0.7 kWh which is correct).

This bar for today goes away at some point -- the screenshots were taken at 8:52, and by 9:01 the bar was gone.

**To Reproduce**

It's just there. I never noticed it before I got my PV installation and therefore started exporting energy, but I'm not sure it wasn't there.

**Expected behavior**

I think the expected behaviour is to only see past days in that graph.

**Screenshots**

Today is Sept. 19:

[image]
[image]

**Hardware information:**
 - Country: Austria
 - Meter: Sagemcom T210-D
 - Encryption enabled: yes
 - AMS reader: ESP32
 - M-bus adapter (if applicable): resistive divider

**Relevant firmware information:**
 - Version: 2.2.21
 - MQTT: yes
 - MQTT payload type: HomeAssistant
 - HAN GPIO: UART2
 - HAN baud and parity: 2400 8E1
 - Temperature sensors n/a
 - ENTSO-E API enabled: yes

**Additional context**
Add any other context about the problem here.

gskjold, 2023-09-19:
Suspect this is some GUI bug where the graph displays "19" in the wrong end. It is data from 19.aug. Will look into

gskjold, 2023-09-27:
I hope this is fixed in my latest build if you want to test: https://github.com/UtilitechAS/amsreader-firmware/issues/641#issuecomment-1736095841

cmock, 2023-09-28:
I installed that build now, but it's hard to say whether it fixes this bug as it doesn't show up consitently...

gskjold, 2023-09-28:
Yeah, I figured it could be a bit random. Let me know if it shows up again

cmock, 2023-10-01:
I now came back to a browser tab that had been open for at least a week, so last reloaded with 2.2.21, and it was showing the bug. 

My AMS is however now running on 2.2.22, so I reloaded the tab and the display was back to normal.

gskjold, 2023-10-02:
Interesting, keep me posted

closed 2023-11-05 by gskjold