UtilitechAS/amsreader-firmware #639: Spikes in L2Current

wknutsen, 2023-09-19:
**Describe the bug**
I found a minor issue in the L2 current data transferred to MQTT. During daytime when the solar panels is producing i have some insane spikes in L2 current data. This is also displayed in frontend aswell. I believe this is related to the solar panels since the issue is only appearing in daytime. I haven't seen this problem on L1 or L3.

**Screenshots**
[image]
[image]

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled: No
 - AMS reader: ESP32S2 Pow-U+
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.21
 - MQTT: Yes
 - MQTT payload type: JSON
 - HAN GPIO: Unknown
 - HAN baud and parity: Uknown
 - Temperature sensors: No
 - ENTSO-E API enabled: Yes

gskjold, 2023-09-19:
Looks like a rollover on a unsigned integer that should have been signed. Will have a look at this

gskjold, 2023-09-20:
I'm not entirely sure what is going on here, from what I can see, data from Norwegian Aidon should be converted correctly. If possible, it would be nice to catch the raw data from the meter when this happens, using debugging:
https://amsleser.no/blog/post/24-telnet-debug

wknutsen, 2023-09-20:
Thank you. I'll look into it next week. This will probably be a pain to find the message. By looking at the telnet screen shoots there is no time stamp. I probably need to log telnet data to file for some hours. Is there any documentation available for Aidon so i can identify where in the message L2 current is located ?

gskjold, 2023-09-21:
Yeah, it is not easy to catch, especially since the telnet times out quite fast. One would have to spot it and telnet right around the time it happens. 

Also, it is possibly fixed in #641

wknutsen, 2023-09-25:
@gskjold 

I can confirm it is not fixed in #641 used that firmware when capturing this.

I was able to capture data from telnet when the error occurred. The interesting part is that the IL2 is not present in the frame.

This is what i have in time series when error IL2 goes sky high.
09/25/2023 3:22:06.538
Export active power  336
L1 Voltage 236.1
L2 Voltage 236.2
L3 Voltage 235.8
L1 Current 1
L2 Current 3.1495E+07
L3 Current 1.3

This is the HDLC frame captured:
(V) HDLC frame:
(V) 7E A1 0B 41  08 83 13 FA  7C E6 E7 00  0F 40 00 00
(V) 00 00 01 0C  
0202 0906 0101 000281FF 0A0B 4149444F4E5F5630303031 
0202 0906 0000 600100FF 0A10 ** 
0202 0906 0000 600107FF 0A04 36353235 
0203 0906 0100 010700FF 06 00000000 0202 0F00 161B 
0203 0906 0100 020700FF 06 00000150 0202 0F00 161B  (336W Active power - (Q2+Q3))
0203 0906 0100 030700FF 06 00000000 0202 0F00 161D 
0203 0906 0100 040700FF 06 00000079 0202 0F00 161D
0203 0906 0100 1F0700FF 10 000A 0202 0FFF 1621     (1.0A IL1 Current phase L1)
0203 0906 0100 470700FF 10 000D 0202 0FFF 1621     (1.3A IL3 Current phase L3)
0203 0906 0100 200700FF 12 0939 0202 0FFF 1623     (236.1V UL1 Phase voltage 4W meter, line voltage 3W meter)
0203 0906 0100 340700FF 12 093A 0202 0FFF 1623     (236.2V UL2 Phase voltage 4W meter, line voltage 3W meter)
0203 0906 0100 480700FF 12 0936 0202 0FFF 1623     (235.8V UL3 Phase voltage 4W meter, line voltage 3W meter)
B1FB 7E

All values in frame matches values in timeseries db except for IL2.

Hard to say whats going on here :/

gskjold, 2023-09-26:
Ah, you are on a IT or TT grid. Then I2 is not present and gets calculated. Very useful info, thanks, I will look at that code

gskjold, 2023-09-26:
Please try the attached firmware:
[esp32s2.zip](https://github.com/UtilitechAS/amsreader-firmware/files/12730925/esp32s2.zip)

closed 2023-10-02 by gskjold