UtilitechAS/amsreader-firmware #640: Tariff peaks missing a peak

wknutsen, 2023-09-19:
**Describe the bug**
Tariff peaks is missing a peak
[image]

I found the value in my timeseries from realtime accumulated hourly
[image]

This is the peaks from my grid company
Dato og tid | Døgnmaks
16.09.2023, kl 00:00 | 4,55 kW
15.09.2023, kl 23:00 | 4,49 kW
03.09.2023, kl 20:00 | 3,40 kW

Could it be that this peak was used on 16th instead of 15th ?

**Hardware information:**
 - Country: Norway
 - Meter: Aidon
 - Encryption enabled: No
 - AMS reader: 
 - M-bus adapter (if applicable):

**Relevant firmware information:**
 - Version: 2.2.21 (Test firmware)
 - MQTT: Yes
 - MQTT payload type: JSON
 - HAN GPIO: Unknown
 - HAN baud and parity: Unknown
 - Temperature sensors: No
 - ENTSO-E API enabled: Yes

ArnieO, 2023-09-20:
Interesting. 
Which hardware are you running?

wknutsen, 2023-09-20:
> Interesting. Which hardware are you running?

I forgot to mention :) ESP32S2 Pow-U+

ArnieO, 2023-09-20:
OK, so I downloaded my own data from  https://elhub.no/elhub-for-sluttbruker/minside/ to check. 
And I confirm there seems to be an issue here:

My 3 highest-consumption hour-intervals according to Elhub are:
[image]
(The CSV file from Elhub needs some "tinkering" in Excel to show it in this format.)

While my GUI shows:
[image]

In my case, I use my meter for testing from time to time - so is a possibility that my amsreader may have not been plugged in or working on 12th sep (which is missing on the graph). 

Therefore:

**CALL FOR HELP**
Has anyone else have seen such discrepancy when comparing with "official data"? 
Norwegian users can log in to "Min side" on Elhub to find the registered data: https://elhub.no/elhub-for-sluttbruker/minside/
Then: 

- Select "Måleverdier" in the top menu
- Click "Last ned måleverdier"
- Select current month (CSV file)
- Import to e.g. Excel, and do whatever is needed to display the data in a sensible way (my Windows installation is in English, and the timestamp date from Elhub is formatted dd/mm/yyyy - which my Excel did not understand correctly)
- Compare with amsleser GUI peak hours

gskjold, 2023-09-20:
This is interesting, I actually noticed on my setup that a peak was missing. Normally I don't see this, because I have a 4.95kwh peak many days in a month, but on this particular setup I knew there should have been two 10ish kwh peaks, but only one is recorded! Wasn't sure if it was a bug or me messing up the data from restore, but now I know

gskjold, 2023-09-21:
Final testing of this feature in issue #641

closed 2023-09-21 by gskjold